robotology/blockfactory #8: Implement support for stateful system

traversaro, 2018-12-23:
This is probably a long term goal, but for now let's collect some possible basic steps: 
* Add support for reading continuous state in `BlockInformation`
* Add support for reading and writing discrete state in `BlockInformation`
* Correctly implement `mdlDerivatives` in the BlockFactory S-Function: https://github.com/robotology/blockfactory/blob/78a8723dd7018f44e6ecc646f5b32868a9fe1f6c/sources/Mex/src/BlockFactory.cpp#L317

traversaro, 2018-12-26:
For the discrete state, it is also important to specify the sample time of the period update connected to the discrete state update. See https://it.mathworks.com/help/simulink/sfg/sample-times-cpp.html#f4-33929 for a  description of this concept in the Simulink docs, and https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_leaf_system.html#a262f71637a37127840acefca3bda8897 for the related API in Drake's System API.

traversaro, 2019-01-20:
Still related to stateful systems, it is relevant to mention that simStateCompliance should be properly set in Simulink: https://github.com/robotology/blockfactory/blob/f8a363e595fb748d313f3785f5a94d831a3ee796/sources/Simulink/src/BlockFactory.cpp#L175 . 
At the moment, it is set to `USE_CUSTOM_SIM_STATE`. This means (see https://it.mathworks.com/help/simulink/sfg/s-function-compliance-with-the-simstate.html) that the S-Function "has mdlGetSimState and mdlSetSimState methods", that are not implemented in the existing BlockFactory S-Function.

diegoferigo, 2019-01-21:
That part of S-Function code is a very old heritage (from first versions of [WBI-Toolbox](https://github.com/robotology-playground/WBI-Toolbox) I think). All of them would need a refreshing, not only this one.

diegoferigo, 2019-11-16:
cc @fjandrad