robotology/blockfactory #17: It is not possible to compile the example in a system without Matlab

traversaro, 2019-01-18:
`BlockFactoryMex` is currently always required in the example (see https://github.com/robotology/blockfactory/blob/master/example/CMakeLists.txt#L35), but it should be required only if Matlab compilation is required (typically using options such as `<package>_USES_MATLAB` or `USES_MATLAB`, or in the case of the example implicitly based on the availability or not of Matlab in the system).

diegoferigo, 2019-01-18:
This case should be handled as I did in the ROS demo:

https://github.com/diegoferigo/blockfactory-demo-ros/blob/9c6bac0723e1fcc479a9e1ed6522c51eafb52feb/CMakeLists.txt#L44-L46

diegoferigo, 2019-01-18:
This case should have been caught from CI, I found a mistake in the script.