robotology/blockfactory #34: When installing the blockfactory and any blockfactory-based plugin in the `/usr` prefix ideally no enviromental variable would need to be set to find the plugins

traversaro, 2019-01-29:
@traversaro wrote:

> Additionally, w.r.t. to debian packaging, it would be good if the default installation prefix was in the search path even without the need to define BLOCKFACTORY_PLUGIN_PATH, so that if in the future we have deb package of BlockFactory + WB-Toolbox, they will just work out of the box without the need to define any env variable.

@diegoferigo replied:

> This is more complicated. If we install plugins in one of the default linker folder, we get the same problems due to `LD_LIBRARY_PATH` we recently discussed in https://github.com/robotology/ycm/issues/199 and https://github.com/robotology/robotology-superbuild/issues/97.

traversaro, 2019-01-29:
> This is more complicated. If we install plugins in one of the default linker folder, we get the same problems due to `LD_LIBRARY_PATH` we recently discussed in https://github.com/robotology/ycm/issues/199 and https://github.com/robotology/robotology-superbuild/issues/97.

I am a bit confused. If when installing in `/usr` on Ubuntu the default installation folder for  blockfactory plugins is `/usr/lib/x86_64-linux-gnu/blockfactory` , why searching it this directory by default there will create any `LD_LIBRARY_PATH` related problem?

traversaro, 2019-01-29:
Anyhow, until we addess packaging this is not really important.

traversaro, 2019-01-29:
Now I perhaps remember that in YARP we discussed in the past to add some kind of logic "search in the same prefix in which YARP it is installed", but we had problems related to how to implement this in a relocatable way, and you needed to obtained the path to the YARP libraries, and this was not possible/easy on Windows.

diegoferigo, 2019-01-29:
Considering that I never had yet direct experience of packaging archives for debian-based distros, I drop here few links that might be useful in the future:

- [Ubuntu: Multiarch Spec](https://wiki.ubuntu.com/MultiarchSpec)
- [Debian: Multiarch Implementation](https://wiki.debian.org/Multiarch/Implementation)
- [SO: What is the proper way to reference shared library plugins with dlopen()?](https://stackoverflow.com/questions/42564706/what-is-the-proper-way-to-reference-shared-library-plugins-with-dlopen)

Probably I didn't explain myself good enough, and probably I might have been wrong. Assuming we are installing the library in `/usr/lib/x86_64-linux-gnu/blockfactory`, I do not honestly know if the linker checks by default all the subfolders inside `/usr/lib/x86_64-linux-gnu`. However, also in this case, including in the deb package a proper conf file installed inside `/etc/ld.so.conf.d/` might suffice.

I'm not sure, but in light of this maybe also with the current situation installing in the system lib directory would not need any edit to the environment variables.