robotology/blockfactory #42: Current status of tunable parameters

diegoferigo, 2019-02-07:
Despite the code generated with Simulink Coder supports tunable parameters, the ones coming from the block masks are not accessible. Refer to the `Operation` selected from the drop-down menu of the `SignalMath` block of the example:

https://github.com/robotology/blockfactory/blob/a05e6d0d192274ebe4f4fae57d8987c9dfa3cde0/example/matlab/AutogenerationExample_grt_rtw/AutogenerationExample.cpp#L104-L107

In line 105 it is hardcoded, and cannot be changed.

Instead, for what concern parameters that come from the workspace (and constants present in the Simulink Model are analogous), they are stored in a static struct:

https://github.com/robotology/blockfactory/blob/a05e6d0d192274ebe4f4fae57d8987c9dfa3cde0/example/matlab/AutogenerationExample_grt_rtw/AutogenerationExample.h#L75-L80

https://github.com/robotology/blockfactory/blob/a05e6d0d192274ebe4f4fae57d8987c9dfa3cde0/example/matlab/AutogenerationExample_grt_rtw/AutogenerationExample.cpp#L390-L400

Despite they can be changed during runtime (the symbol is static), it's basically impossible to know what is the index of the parameter you want to access. In this case there's only one, but if there were more, the cpp will contain a curly braces initialization for all of them. It would be great having a way to access these parameters e.g. by name (and I mean string, not name of the variable like `P_AutogenerationExample_T_.Constant_Value` in this case), but the approach used by the coder does not allow it.

cc @traversaro

diegoferigo, 2019-02-07:
Also related to https://github.com/robotology/whole-body-controllers/issues/26

traversaro, 2019-02-07:
Are we generating the C API to access the parameters, as described in https://github.com/Kvixen/Simulix#exporting-your-model-from-simulink ?

diegoferigo, 2019-02-07:
I remember I tried them but I had no luck even with them.