robotology/blockfactory #44: Compilation failed with MATLAB 2018a on Windows 10 Version 1803 with CMake 3.12.0

bemilio, 2019-05-17:
Compiling BlockFactory with MATLAB 2018a on Windows 10 Version 1803 with CMake 3.12.0 we got the following error: 
~~~
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Tools\MSVC\14.14.26428\bin\HostX86\x64\link.exe /ERRORREPORT:QUEUE /OUT:"C:\dev\git\robotology-superbuild\build\robotology\BlockFactory\bin\Release\BlockFactorySimulinkCoder.dll" /INCREMENTAL:NO /NOLOGO /LIBPATH:"C:\dev\vcpkg\installed\x64-windows\lib" /LIBPATH:"C:\dev\vcpkg\installed\x64-windows\lib\manual-link" ..\..\lib\Release\BlockFactoryCore.lib ..\..\lib\Release\shlibpp.lib kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib "C:\dev\vcpkg\installed\x64-windows\lib\*.lib" /DEF:"C:/dev/git/robotology-superbuild/build/robotology/BlockFactory/sources/SimulinkCoder/SimulinkCoder.dir/Release/exports.def" /MANIFEST /MANIFESTUAC:"level='asInvoker' uiAccess='false'" /manifest:embed /PDB:"C:/dev/git/robotology-superbuild/build/robotology/BlockFactory/bin/Release/BlockFactorySimulinkCoder.pdb" /SUBSYSTEM:CONSOLE /TLBID:1 /DYNAMICBASE /NXCOMPAT /IMPLIB:"C:/dev/git/robotology-superbuild/build/robotology/BlockFactory/lib/Release/BlockFactorySimulinkCoder.lib" /MACHINE:X64  /machine:x64 /DLL SimulinkCoder.dir\Release\CoderBlockInformation.obj
     Creating library C:/dev/git/robotology-superbuild/build/robotology/BlockFactory/lib/Release/BlockFactorySimulinkCoder.lib and object C:/dev/git/robotology-superbuild/build/robotology/BlockFactory/lib/Release/BlockFactorySimulinkCoder.exp
  SimulinkCoder.vcxproj -> C:\dev\git\robotology-superbuild\build\robotology\BlockFactory\bin\Release\BlockFactorySimulinkCoder.dll
AppLocalFromInstalled:
  C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -ExecutionPolicy Bypass -noprofile -File "C:\dev\vcpkg\scripts\buildsystems\msbuild\applocal.ps1" "C:\dev\git\robotology-superbuild\build\robotology\BlockFactory\bin\Release\BlockFactorySimulinkCoder.dll" "C:\dev\vcpkg\installed\x64-windows\bin" "SimulinkCoder.dir\Release\SimulinkCoder.tlog\SimulinkCoder.write.1u.tlog" "SimulinkCoder.dir\Release\vcpkg.applocal.log"
FinalizeBuildStatus:
  Deleting file "SimulinkCoder.dir\Release\SimulinkCoder.tlog\unsuccessfulbuild".
  Touching "SimulinkCoder.dir\Release\SimulinkCoder.tlog\SimulinkCoder.lastbuildstate".
Done Building Project "C:\dev\git\robotology-superbuild\build\robotology\BlockFactory\sources\SimulinkCoder\SimulinkCoder.vcxproj" (default targets).
Done Building Project "C:\dev\git\robotology-superbuild\build\robotology\BlockFactory\ALL_BUILD.vcxproj" (default targets) -- FAILED.

Build FAILED.

"C:\dev\git\robotology-superbuild\build\robotology\BlockFactory\ALL_BUILD.vcxproj" (default target) (1) ->
"C:\dev\git\robotology-superbuild\build\robotology\BlockFactory\sources\Simulink\Simulink.vcxproj" (default target) (5) ->
(Link target) ->
  LINK : error LNK2001: unresolved external symbol mexCreateMexFunction [C:\dev\git\robotology-superbuild\build\robotology\BlockFactory\sources\Simulink\Simulink.vcxproj]
  LINK : error LNK2001: unresolved external symbol mexDestroyMexFunction [C:\dev\git\robotology-superbuild\build\robotology\BlockFactory\sources\Simulink\Simulink.vcxproj]
  LINK : error LNK2001: unresolved external symbol mexFunctionAdapter [C:\dev\git\robotology-superbuild\build\robotology\BlockFactory\sources\Simulink\Simulink.vcxproj]
  LINK : error LNK2001: unresolved external symbol mexfilerequiredapiversion [C:\dev\git\robotology-superbuild\build\robotology\BlockFactory\sources\Simulink\Simulink.vcxproj]
  C:/dev/git/robotology-superbuild/build/robotology/BlockFactory/lib/Release/BlockFactory.lib : fatal error LNK1120: 4 unresolved externals [C:\dev\git\robotology-superbuild\build\robotology\BlockFactory\sources\Simulink\Simulink.vcxproj]

    0 Warning(s)
    5 Error(s)

Time Elapsed 00:00:08.46
~~~

traversaro, 2019-05-17:
cc @diegoferigo 

I guess this can be related to https://github.com/robotology/blockfactory/issues/4 .

traversaro, 2019-05-17:
The only related issue that I found online is  https://www.reddit.com/r/fictrac/comments/bojxgg/error_installing_nloptx64windows_using_vcpkg/ .

diegoferigo, 2019-05-17:
Yes it is related to #4. Blockfactory ships a vendored version of `FindMatlab.cmake`, I would try to rename that file from the `cmake` folder and check what happens with the default version. `3.12` should be recent enough to have the latest fixes on symbol visibility (I should double check though).

I suspect the the `FindMatlab.cmake` does not handle as it should the `2018a` version. Unfortunately keeping track of all Matlab version is quite demanding. If with the upstream version still doesn't work, we should fix a f2f session.

traversaro, 2019-05-17:
I suspect the issue is in the CMake's upstream version of `FindMatlab` module. We experienced the same problem in https://github.com/robotology-dependencies/qpOASES, that recently vendored the latest version of `FindMatlab.cmake` from CMake's upstream, see https://github.com/robotology-dependencies/qpOASES/pull/11 .

traversaro, 2019-05-17:
The problem seems that "something" is looking for the `mexCreateMexFunction` and  the other functions referenced in the linker error, but those functions are not defined. With @bemilio I briefly tried to find that in some part of the code, but I was unable to find this symbol referenced in any C/C++ file. After checking a bit the recent history of the `FindMatlab.cmake` file, I think that these errors can be related to the fact that all those functions are explicitly exported in https://github.com/robotology/blockfactory/blob/master/cmake/FindMatlab.cmake#L1031 (see https://gitlab.kitware.com/cmake/cmake/issues/17971 for the related CMake's MR). I think it would be nice to comment the  assignment of link_flags in https://github.com/robotology/blockfactory/blob/master/cmake/FindMatlab.cmake#L1031, and see how the error changes.

traversaro, 2019-05-17:
Another test I would do is to uncomment the commented assign in https://github.com/robotology/blockfactory/blob/master/cmake/FindMatlab.cmake#L962 and see if that fix something.

diegoferigo, 2019-05-20:
I had a look today on the setup and I didn't mange to solve the problem. I was suspecting that somehow the following section was not executed:

https://github.com/robotology/blockfactory/blob/a05e6d0d192274ebe4f4fae57d8987c9dfa3cde0/cmake/FindMatlab.cmake#L1025-L1033

But instead it is. Also `Matlab_HAS_CPP_API` is set, and the symbols should be properly exported. However the linker complains.

I also checked if upstream there is an updated version of `FindMatlab.cmake`. There are minor differences wrt our vendored file, nothing related to this section though.

diegoferigo, 2019-05-20:
Commenting out the lines that force the export of those symbols solves the linking error. However, there is a problem with:

https://github.com/robotology/blockfactory/blob/a05e6d0d192274ebe4f4fae57d8987c9dfa3cde0/sources/Core/include/BlockFactory/Core/BlockInformation.h#L31

diegoferigo, 2019-05-20:
Ok I managed to track down the problem (there are two independent problems).

**Problem 1: `FindMatlab.cmake`:** Commenting out [this block](https://github.com/robotology/blockfactory/blob/a05e6d0d192274ebe4f4fae57d8987c9dfa3cde0/cmake/FindMatlab.cmake#L1030-L1033) fixes the linking problem.

**Problem 2: extern symbol:** Changing the following line

https://github.com/robotology/blockfactory/blob/a05e6d0d192274ebe4f4fae57d8987c9dfa3cde0/sources/Core/include/BlockFactory/Core/BlockInformation.h#L31

to

```cpp
const std::string BlockOptionPrioritizeOrder = "wbt.BlockOptionPrioritizeOrder";
```

fixes the complaints about the extern symbol. Using inline does not work because the compiler would require C++17.

bemilio, 2019-05-20:
It looks like updating CMake from 3.12.0 to 3.14.4 has fixed the last (generic) error 

> CMake Error : error : generator : Visual Studio 15 2017 Win64

Still, the output of the in Windows compiler is weird and it does not show anything like "build completed with success". I hope it's fine.

traversaro, 2019-05-20:
The `Matlab_HAS_CPP_API` flag seems to be fishy. It is enabled by default for Matlab & Matlab Runtime versions >= 9.4, but for some tests with Matlab Runtime  2018a (see https://github.com/robotology-dependencies/qpOASES/issues/12) I had to set it to manually set it to 0. I suspect that the issue is in CMake's upstream in FindMatlab, see https://github.com/robotology-dependencies/qpOASES/issues/12#issuecomment-493788107 for more info.

traversaro, 2019-06-14:
Related upstream (CMake) issue:  https://gitlab.kitware.com/cmake/cmake/issues/19382 .

diegoferigo, 2019-07-26:
@RAR1989 @isorrentino Can you please check if on Windows the https://github.com/diegoferigo/blockfactory/tree/fix/windows branch compiles?

PR #54 (2020-02-01): Fix compilation on Windows
Tested on my laptop  with Matlab R2019b . 

Fixes #44 and robotology/robotology-superbuild#226.

Remake of https://github.com/robotology/blockfactory/pull/45 with the additional CMake fixes to fix https://gitlab.kitware.com/cmake/cmake/issues/19382, that was the root cause for the link failure.
closed 2020-02-01 by diegoferigo