robotology/blockfactory #46: Compilations fail with Matlab2019b on Macos

lrapetti, 2019-09-27:
Compilation is failing with the following error message:
```
$ make
[  9%] Built target mxpp
[ 28%] Built target shlibpp
[ 71%] Built target Core
[ 80%] Built target SimulinkCoder
[ 85%] Linking CXX shared library ../../lib/BlockFactory.mexmaci64
Undefined symbols for architecture x86_64:
  "_mexfilerequiredapiversion", referenced from:
     -exported_symbol[s_list] command line option
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/BlockFactory.mexmaci64] Error 1
make[1]: *** [sources/Simulink/CMakeFiles/Simulink.dir/all] Error 2
make: *** [all] Error 2
```

lrapetti, 2019-09-27:
Possibly duplicated of https://github.com/robotology/blockfactory/issues/4

traversaro, 2019-09-27:
We probably need to bump up the request that I made in https://it.mathworks.com/matlabcentral/answers/377799-compiling-mex-files-without-the-mex-command to MathWorks about detailed compilation instructions also on macOS, so that we can update the CMake code to compile mex extensions. To increase its visibility, it would be great if someone could open a new question in https://it.mathworks.com/matlabcentral/answers .

lrapetti, 2019-09-27:
The problem was fixed by https://github.com/robotology/blockfactory/issues/4#issuecomment-460187364