robotology/blockfactory #47: Compilation failure on Windows without Matlab installed

traversaro, 2019-11-13:
While setting up the Windows CI of the robotology-superbuild, I am experiencing this error: 
~~~
2019-11-13T15:10:26.6676384Z   Checking Build System
2019-11-13T15:10:26.8114995Z   Creating directories for 'BlockFactory'
2019-11-13T15:10:26.9428104Z   Building Custom Rule D:/a/robotology-superbuild/robotology-superbuild/CMakeLists.txt
2019-11-13T15:10:26.9669757Z   Performing download step (git clone) for 'BlockFactory'
2019-11-13T15:10:27.0045007Z   Cloning into 'BlockFactory'...
2019-11-13T15:10:27.5160371Z   Already on 'master'
2019-11-13T15:10:27.5161191Z   Your branch is up to date with 'origin/master'.
2019-11-13T15:10:28.1069260Z   Performing update step for 'BlockFactory'
2019-11-13T15:10:28.3803191Z   Current branch master is up to date.
2019-11-13T15:10:28.9344208Z   No patch step for 'BlockFactory'
2019-11-13T15:10:28.9712795Z   Performing configure step for 'BlockFactory'
2019-11-13T15:10:28.9872562Z   Not searching for unused variables given on the command line.
2019-11-13T15:10:28.9902568Z   loading initial cache file D:/a/robotology-superbuild/robotology-superbuild/build/robotology/BlockFactory/CMakeFiles/YCMTmp/BlockFactory-cache-Debug.cmake
2019-11-13T15:10:30.1316248Z   -- The CXX compiler identification is MSVC 19.23.28106.4
2019-11-13T15:10:30.3629641Z   -- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Enterprise/VC/Tools/MSVC/14.23.28105/bin/Hostx64/x64/cl.exe
2019-11-13T15:10:31.6319293Z   -- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Enterprise/VC/Tools/MSVC/14.23.28105/bin/Hostx64/x64/cl.exe -- works
2019-11-13T15:10:31.6344818Z   -- Detecting CXX compiler ABI info
2019-11-13T15:10:32.8949280Z   -- Detecting CXX compiler ABI info - done
2019-11-13T15:10:32.8983496Z   -- Detecting CXX compile features
2019-11-13T15:10:32.8995940Z   -- Detecting CXX compile features - done
2019-11-13T15:10:33.1586860Z   -- Could NOT find Matlab (missing: Matlab_INCLUDE_DIRS Matlab_MEX_LIBRARY Matlab_MEX_EXTENSION Matlab_ROOT_DIR Matlab_MX_LIBRARY MX_LIBRARY ENG_LIBRARY MAIN_PROGRAM SIMULINK) (found version "NOTFOUND")
2019-11-13T15:10:33.2308553Z   -- Configuring done
2019-11-13T15:10:33.3369794Z   -- Generating done
2019-11-13T15:10:33.3398619Z   -- Build files have been written to: D:/a/robotology-superbuild/robotology-superbuild/build/robotology/BlockFactory
2019-11-13T15:10:33.3675435Z   Performing build step for 'BlockFactory'
2019-11-13T15:10:33.4514732Z   Microsoft (R) Build Engine version 16.3.2+e481bbf88 for .NET Framework
2019-11-13T15:10:33.4515253Z   Copyright (C) Microsoft Corporation. All rights reserved.
2019-11-13T15:10:33.4515586Z   
2019-11-13T15:10:33.8272750Z     Checking Build System
2019-11-13T15:10:33.9919084Z     Building Custom Rule D:/a/robotology-superbuild/robotology-superbuild/robotology/BlockFactory/deps/sharedlibpp/src/CMakeLists.txt
2019-11-13T15:10:34.1265901Z     version.cpp
2019-11-13T15:10:34.3974557Z     SharedLibrary.cpp
2019-11-13T15:10:35.7431655Z     SharedLibraryFactory.cpp
2019-11-13T15:10:36.1144519Z D:\a\robotology-superbuild\robotology-superbuild\robotology\BlockFactory\deps\sharedlibpp\src\SharedLibraryFactory.cpp(227,1): warning C4996: 'getenv': This function or variable may be unsafe. Consider using _dupenv_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details. [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\BlockFactory\deps\sharedlibpp\src\shlibpp.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\BlockFactory.vcxproj]
2019-11-13T15:10:36.2704871Z     Generating Code...
2019-11-13T15:10:36.4058779Z   C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdlib.h(1191): message : see declaration of 'getenv' [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\BlockFactory\deps\sharedlibpp\src\shlibpp.vcxproj]
2019-11-13T15:10:36.4391075Z     Auto build dll exports
2019-11-13T15:10:36.6175692Z        Creating library D:/a/robotology-superbuild/robotology-superbuild/build/robotology/BlockFactory/lib/Debug/shlibppd.lib and object D:/a/robotology-superbuild/robotology-superbuild/build/robotology/BlockFactory/lib/Debug/shlibppd.exp
2019-11-13T15:10:36.7528705Z     shlibpp.vcxproj -> D:\a\robotology-superbuild\robotology-superbuild\build\robotology\BlockFactory\bin\Debug\shlibppd.dll
2019-11-13T15:10:37.3485863Z     Building Custom Rule D:/a/robotology-superbuild/robotology-superbuild/robotology/BlockFactory/sources/Core/CMakeLists.txt
2019-11-13T15:10:37.4524846Z     Block.cpp
2019-11-13T15:10:37.9820992Z     BlockInformation.cpp
2019-11-13T15:10:38.2825423Z     Log.cpp
2019-11-13T15:10:38.8226196Z     Parameter.cpp
2019-11-13T15:10:39.0936118Z     Parameters.cpp
2019-11-13T15:10:40.2583101Z     ConvertStdVector.cpp
2019-11-13T15:10:40.6957546Z C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.23.28105\include\xmemory(758,82): error C2220: the following warning is treated as an error [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\BlockFactory\sources\Core\Core.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\BlockFactory.vcxproj]
2019-11-13T15:10:40.7016614Z   C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.23.28105\include\xmemory(1581): message : see reference to function template instantiation 'void std::_Default_allocator_traits<_Alloc>::construct<_Ty,const double&>(_Alloc &,_Objty *const ,const double &)' being compiled [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\BlockFactory\sources\Core\Core.vcxproj]
2019-11-13T15:10:40.7051818Z   C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.23.28105\include\xmemory(1581): message :         with [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\BlockFactory\sources\Core\Core.vcxproj]
2019-11-13T15:10:40.7064525Z   C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.23.28105\include\xmemory(1581): message :         [ [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\BlockFactory\sources\Core\Core.vcxproj]
2019-11-13T15:10:40.7071003Z   C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.23.28105\include\xmemory(1581): message :             _Alloc=std::allocator<int>, [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\BlockFactory\sources\Core\Core.vcxproj]
2019-11-13T15:10:40.7077241Z   C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.23.28105\include\xmemory(1581): message :             _Ty=int, [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\BlockFactory\sources\Core\Core.vcxproj]
2019-11-13T15:10:40.7083605Z   C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.23.28105\include\xmemory(1581): message :             _Objty=int [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\BlockFactory\sources\Core\Core.vcxproj]
2019-11-13T15:10:40.7089752Z   C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.23.28105\include\xmemory(1581): message :         ] [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\BlockFactory\sources\Core\Core.vcxproj]
2019-11-13T15:10:40.7096281Z   C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.23.28105\include\xmemory(1581): message : see reference to function template instantiation 'void std::_Default_allocator_traits<_Alloc>::construct<_Ty,const double&>(_Alloc &,_Objty *const ,const double &)' being compiled [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\BlockFactory\sources\Core\Core.vcxproj]
2019-11-13T15:10:40.7111039Z   C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.23.28105\include\xmemory(1581): message :         with [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\BlockFactory\sources\Core\Core.vcxproj]
2019-11-13T15:10:40.7117078Z   C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.23.28105\include\xmemory(1581): message :         [ [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\BlockFactory\sources\Core\Core.vcxproj]
[... cut ...]
2019-11-13T15:10:41.6572444Z D:\a\robotology-superbuild\robotology-superbuild\robotology\BlockFactory\sources\Core\src\FactorySingleton.cpp(29,1): warning C4996: 'getenv': This function or variable may be unsafe. Consider using _dupenv_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details. [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\BlockFactory\sources\Core\Core.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\BlockFactory.vcxproj]
~~~

diegoferigo, 2019-11-13:
I suspect that you are compiling the code in `Debug`, right? In this mode, blockfactory treats many warnings as errors. Windows has never been tested thoroughly, and I suspect that new versions of the compiler introduced new warnings that are not ignored.

You can refer to this chunk for more details:

https://github.com/robotology/blockfactory/blob/a05e6d0d192274ebe4f4fae57d8987c9dfa3cde0/cmake/TargetCompileWarnings.cmake#L40-L44

traversaro, 2019-11-14:
> Windows has never been tested thoroughly, and I suspect that new versions of the compiler introduced new warnings that are not ignored.

Yes, that is exactly why I typically prefer to avoid compiling with Warning as Errors by default. However, to disable this behavior optionally, can we transform `TREAT_WARNINGS_AS_ERRORS` in a proper CMake option? It has already a option-like face, but as it is not saved in cache it may be difficult to manage.

traversaro, 2019-11-14:
For the time being, I disabled directly the ENABLE_WARNING options in https://github.com/robotology/robotology-superbuild/pull/296/commits/a49a7b9f150aa806f6a2d23ebafb1decf8447317 .

diegoferigo, 2019-11-15:
I managed to replicate the windows problem, see logs [here](https://github.com/diegoferigo/blockfactory/runs/304587104). I am preparing a fix.

closed 2024-09-24 by traversaro