robotology/blockfactory #50: Specifying sample times

bemilio, 2019-12-20:
I was wondering if and how it is possible to specify the sample time at which a blockfactory Simulink block is executed

@diegoferigo @traversaro

bemilio, 2019-12-20:
APparently it was pretty simple - I just had to put the `rateTransition` blocks before and after the block with the given sampling time. I don't know if there is a parameter (or if there would make sense to have one) to specify it directly in the block.

diegoferigo, 2019-12-20:
@bemilio Currently blockfactory allows implementing unit blocks that provide an instantaneous output (if needed, with an internal memory). Under this assumption, the sample time of the simulation is not passed to the blocks, and there isn't yet an unified way to distribute this information. There's a specific issue that tracks specifically this requirement https://github.com/robotology/blockfactory/issues/8.

There are cases where we need to know the sample time (e.g. the real-time synchronizer block). In that case we pass the sampling time as a mask parameter. It goes under the assumption that blockfactory only support models simulated with a fixed step.

> Apparently it was pretty simple - I just had to put the rateTransition blocks before and after the block with the given sampling time.

Your solution is not very clear to me, can you argument it a bit more in detail?

bemilio, 2019-12-20:
here are two screenshots of the setup (one is just a zoom). Blocks and signals have different colors depending on the sample time at which they run.

The yellow blocks are the `rate transition` Simulink blocks. Given a specific output sample rate, they propagate that sample rate and manage the signal sampling frequency transition.

[image]
[image]

diegoferigo, 2019-12-27:
@bemilio I think you are among the first ones to use different sampling rate within the same model (maybe with @CarlottaSartore). I was not aware of the existence of those blocks for the transition.

As far as I knew, an alternative solution is to put all the blocks with a different sampling rate in another subsystem, and then configure the subsystem's step size. Probably in this way you don't need any additional blocks. Though, I'm not sure what is their actual effect.

bemilio, 2020-01-03:
> @bemilio I think you are among the first ones to use different sampling rate within the same model (maybe with @CarlottaSartore). I was not aware of the existence of those blocks for the [...]

Adding to that, I noticed that Simulink has some problems setting the dimensionality of the signals when using these `rateTransition` blocks. I suggest setting the `initial condition` parameter of these blocks to a value with a coherent dimension if you have trouble compiling.

bemilio, 2020-01-03:
I think I can close this, the solution seems to work

closed 2020-01-03 by bemilio