robotology/blockfactory #51: Suggestion for configure Size And Ports into Initialize phase

DavideAntonucci, 2019-12-24:
Hi all, 

My colleagues (@liesrock @alaurenzi) and I are working with the blockfactory to use our framework in Matlab/Simulink.

Is there a possibility to declare a size port during the initialize phase?

Since I use the configureSizeAndPorts method to declare my inputs and outputs with a dynamic size port and I know the size after the initialize phase, I was wondering if there is a chance to do it in this phase.

Davide.

diegoferigo, 2019-12-27:
Hi @DavideAntonucci 

TL;DR
No, if you want to fix signal sizes, you have to know them in the `configureSizeAndPorts`.

> Is there a possibility to declare a size port during the initialize phase?
>
> Since I use the configureSizeAndPorts method to declare my inputs and outputs with a dynamic size port and I know the size after the initialize phase, I was wondering if there is a chance to do it in this phase.

Before running a simulation, Simulink performs a series of predefined steps to prepare the DAG that describes the execution order of the blocks. You can read more [at this link](https://it.mathworks.com/help/simulink/sfg/how-the-simulink-engine-interacts-with-c-s-functions.html).

To answer your question, you need to know that the `Block` methods relevant during initialization are mapped as follows with the Simulink Engine:

- `mdlInitializeSizes` -> `Block::configureSizeAndPorts`
- `mdlStart` -> `Block::initialize`

These two steps are completely independent. This means that, as documented, all the block resources configured / allocated in `configureSizeAndPorts` get deleted. In fact, a temporary `Block` object [is created](https://github.com/robotology/blockfactory/blob/a05e6d0d192274ebe4f4fae57d8987c9dfa3cde0/sources/Simulink/src/BlockFactory.cpp#L111-L113) and destroyed afterwards. What's important is that in this phase Simulink performs something similar:

1. Get all the ports with fixed-size signals and checks for incompatibilities
1. Get all the dynamic signals and tries to propagate / guess signal size
1. Creates the DAG with the resulting block connections and their signal size

If Simulink cannot figure out the signal size, it fails with an error and does not continue. This is not so unlikely if you're writing your own blocks. In fact, if you overuse dynamic size, it might happen that Simulink is not able to finalize the size propagation.

This is more or less what happens under the hood in the first phase of the simulation, or also when you press Ctrl+D. 

Despite it is not forbidden to call in the `mdlStart` phase the methods that operate on signal sizes, if I recall well, they do not work (and no warning / error is issued). The next `mdlStart` phase expects to have already the information about the signal size and the only thing you might want to do in this step is to add few checks that sizes are what expected (example [QpOases.cpp](https://github.com/robotology/wb-toolbox/blob/bbd3e52ce53a705498537bc99410c59619c46fb2/toolbox/library/src/QpOases.cpp#L254-L257)).

alaurenzi, 2019-12-30:
Hi @diegoferigo, thanks for your time. This would not be a problem for us, but, if I understood correctly from @DavideAntonucci, the `mdlInitializeSizes` (implying `Block::configureSizeAndPorts`) is also called when a new block is dropped to the simulink panel. Maybe this is related to the following note in the Simulink documentation that you linked before:

```
Note
The mdlInitializeSizes callback method also runs when you enter the name of a compiled S-function into the S-Function Block Parameters dialog box.
```

Now, when a block is dropped, we have no idea about the ports size, because they depend on the URDF which, in out context, could not be available until the play button is pressed. Have you experienced this problem as well?

traversaro, 2019-12-30:
> Now, when a block is dropped, we have no idea about the ports size, because they depend on the URDF which, in out context, could not be available until the play button is pressed. Have you experienced this problem as well?

I guess @diegoferigo can answer with more details, but in the meanwhile I think I can answer that in our case the URDF and the list of controller joints is assumed to be known at that stage, as you can see in the case of a block such as the one that computes the Jacobian: 
* https://github.com/robotology/wb-toolbox/blob/f110c4b641a09a4bb392161db2ab79d0afc0adbe/toolbox/library/src/Jacobian.cpp#L93
* https://github.com/robotology/wb-toolbox/blob/f110c4b641a09a4bb392161db2ab79d0afc0adbe/toolbox/base/src/WBBlock.cpp#L210

diegoferigo, 2020-01-02:
> Maybe this is related to the following note in the Simulink documentation that you linked before:
>>The mdlInitializeSizes callback method also runs when you enter the name of a compiled S-function into the S-Function Block Parameters dialog box.

It could be, I think it is used to update the mask I/O appearance as soon as you drop the block in a model.

> Now, when a block is dropped, we have no idea about the ports size, because they depend on the URDF which, in out context, could not be available until the play button is pressed. Have you experienced this problem as well?

As @traversaro wrote, blocks belonging to our WBToolbox have access to the URDF already from the `configureSizeAndPorts` method. In this way we can avoid using ports with dynamic size. This is particularly important because blocks that read robot measurements (e.g. joint positions) are typically input blocks. Fixing the size of their output signal simplifies a lot the process of understanding signal sizes of the overall Simulink model.

In your case, if the URDF is only available when you press the play button, means that you have access to its data starting from the `mdlStart` / `Block::initialize` phase. How do you pass the URDF file to the block? As a mask parameter?

Right now I don't see any solution that does not involve moving the URDF parsing to the `configureSizeAndPorts` step, unless you find a way to handle your problem with dynamically-sized ports.

traversaro, 2020-01-03:
The following Simulink functions to define "accepted" values for dynamically sized ports seems to be interesting:

* `mdlSetInputPortDimensionInfo` : https://www.mathworks.com/help/simulink/sfg/mdlsetinputportdimensioninfo.html 
* `mdlSetOutputPortDimensionInfo` : https://www.mathworks.com/help/simulink/sfg/mdlsetoutputportdimensioninfo.html

depending on when they are called, implementing in some way support for calling them in BlockFactory could solve this problem. Anyone has experience with them? 
Another interesting function is `mdlSetDefaultPortDimensionInfo` ( https://www.mathworks.com/help/simulink/sfg/mdlsetdefaultportdimensioninfo.html ).

An interesting project (similar to BlockFactory) that implements them is [EasyLink](https://github.com/gjlaurent/easylink), see: 
https://github.com/gjlaurent/easylink/blob/8aecaf915a827522aa73d0f64b894810849570e9/include/sfunDefinitions.h#L32

diegoferigo, 2020-01-03:
@traversaro I remember I played a bit with those methods in order to solve propagation problems we faced in the past, without any luck unfortunately. I recall that I couldn't make them work as expected, not sure if it was my fault of using them in the wrong way.

> An interesting project (similar to BlockFactory) that implements them is EasyLink

This is extremely interesting. We should have a look if they support also variable step solvers and other cases we did not yet implement.

traversaro, 2020-01-03:
> @traversaro I remember I played a bit with those methods in order to solve propagation problems we faced in the past, without any luck unfortunately.

I am not sure, but I think the propagation problems that you had are not exactly the same as just declaring some port as dynamic and specifying their values after the initialization, as discussed in this issue.

traversaro, 2020-01-27:
This discussion can be related: https://github.com/robotology/wb-toolbox/issues/97 .

DavideAntonucci, 2020-01-28:
Hi all,

About this issue, we found a possible solution that consist to Configure Size and port in this way:

Since we need to have the ROS Master and XBotCore Node alive to get, for instance, the joint number,  during the configure size and port phase we check:

1) ROS Master and XBotCore Node alive.
    - Yes: get joint number.
    - No: joint number equal to one.
2) Print only the errors related of the block like parameters etc...

Then if the first step that it's checked by the Initialize phase too, is not satisfied, the Initialize method will print the error.

This allows us to develop the libraries without having errors, as soon as the play button is pressed, all checks will be performed.