robotology/blockfactory #58: Update CI to Github Actions

diegoferigo, 2020-11-13:


traversaro, 2021-03-11:
This should be much simpler now that we have conda-provided binaries (see for example https://github.com/dic-iit/bipedal-locomotion-framework/pull/221) and MATLAB libraries to compile on GitHub Action (see for example https://github.com/robotology/idyntree/pull/839) this should be much simpler. cc @nunoguedelha this could be a good task if you want to get used with this technologies.

diegoferigo, 2021-03-12:
Maybe updating https://github.com/robotology/blockfactory/pull/48 that was almost done? I'm not sure why it got stuck after approval.

traversaro, 2021-03-12:
> Maybe updating #48 that was almost done? I'm not sure why it got stuck after approval.

From my point of view that can be merged even right away.

diegoferigo, 2021-03-12:
Since a lot of time has passes, before merging maybe it would be better to re-run the workflow to validate if still works.

closed 2025-01-30 by diegoferigo