robotology/blockfactory #68: CMake Error related to MatLab Libraries  with MATLAB 2021a/2022a on Windows 10 Version 21H2 with CMake 3.22.3

kevinwidjaja21, 2022-03-20:
So, I have been trying to compile BlockFactory with Matlab R2021a and keep getting the following error.

`PS F:\simconnect-monitor\blockfactory> cmake -S . -B build
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.19044.
-- Could NOT find Matlab (missing: Matlab_MEX_LIBRARY Matlab_MX_LIBRARY ENG_LIBRARY) (found version "9.10")
CMake Error at C:/Program Files/CMake/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Matlab (missing: Matlab_MEX_LIBRARY Matlab_MX_LIBRARY
  MX_LIBRARY) (found version "9.10")
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  cmake/FindMatlab.cmake:1872 (find_package_handle_standard_args)
  deps/mxpp/CMakeLists.txt:45 (find_package)

-- Configuring incomplete, errors occurred!
See also "F:/simconnect-monitor/blockfactory/build/CMakeFiles/CMakeOutput.log".`

I have already tried also with Matlab R2022a (which I notice is not listed by FindMatlab.cmake so I installed R2021a. Still haven't uninstalled R2022a). I am on Windows 10 and have tried to add _matlabr2021aroot_extern\lib\win64\microsoft to the PATH environment variable.

Sorry if the question seems trivial but I don't know what to do.

traversaro, 2022-03-20:
Hi @kevinwidjaja21 ! I am away from keyboard now, but two things to check are:

* Is the matlab command working in any command prompt?
* If the automatic detection of CMake is not working, you can manually specify the location of Matlab via the Matlab_ROOT_DIR cmake variable.

kevinwidjaja21, 2022-03-21:
Hi @traversaro . So, I have tried adding `set(Matlab_ROOT_DIR "D:/MATLAB/R2021a")` to FindMatlab.cmake just after `include(CheckCCompilerFlag)`. I also tried to add `-D Matlab_ROOT_DIR="D:/MATLAB/R2021a"`. No difference, still the same issue. 
I have confirmed that it is are the correct path to the MatLab root and I confirmed that I can launch Matlab through PowerShell, command prompt, and Git Bash.

kevinwidjaja21, 2022-03-23:
Ok, this is what's inside CMakeCache.txt 
```cmake
# This is the CMakeCache file.
# For build in directory: f:/simconnect-monitor/blockfactory/build
# It was generated by CMake: C:/Program Files/CMake/bin/cmake.exe
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Compile BlockFactory as a shared library
BUILD_SHARED_LIBS:BOOL=ON

//Create tests using CMake
BUILD_TESTING:BOOL=OFF

//Path to a program.
CMAKE_AR:FILEPATH=C:/Program Files (x86)/Microsoft Visual Studio/2017/BuildTools/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/lib.exe

//Semicolon separated list of supported configuration types, only
// supports Debug, Release, MinSizeRel, and RelWithDebInfo, anything
// else will be ignored.
CMAKE_CONFIGURATION_TYPES:STRING=Debug;Release;MinSizeRel;RelWithDebInfo

//Flags used by the CXX compiler during all build types.
CMAKE_CXX_FLAGS:STRING=/DWIN32 /D_WINDOWS /W3 /GR /EHsc

//Flags used by the CXX compiler during DEBUG builds.
CMAKE_CXX_FLAGS_DEBUG:STRING=/MDd /Zi /Ob0 /Od /RTC1

//Flags used by the CXX compiler during MINSIZEREL builds.
CMAKE_CXX_FLAGS_MINSIZEREL:STRING=/MD /O1 /Ob1 /DNDEBUG

//Flags used by the CXX compiler during RELEASE builds.
CMAKE_CXX_FLAGS_RELEASE:STRING=/MD /O2 /Ob2 /DNDEBUG

//Flags used by the CXX compiler during RELWITHDEBINFO builds.
CMAKE_CXX_FLAGS_RELWITHDEBINFO:STRING=/MD /Zi /O2 /Ob1 /DNDEBUG

//Libraries linked by default with all C++ applications.
CMAKE_CXX_STANDARD_LIBRARIES:STRING=kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib

//Flags used by the linker during all build types.
CMAKE_EXE_LINKER_FLAGS:STRING=/machine:X86

//Flags used by the linker during DEBUG builds.
CMAKE_EXE_LINKER_FLAGS_DEBUG:STRING=/debug /INCREMENTAL

//Flags used by the linker during MINSIZEREL builds.
CMAKE_EXE_LINKER_FLAGS_MINSIZEREL:STRING=/INCREMENTAL:NO

//Flags used by the linker during RELEASE builds.
CMAKE_EXE_LINKER_FLAGS_RELEASE:STRING=/INCREMENTAL:NO

//Flags used by the linker during RELWITHDEBINFO builds.
CMAKE_EXE_LINKER_FLAGS_RELWITHDEBINFO:STRING=/debug /INCREMENTAL

//User executables (bin)
CMAKE_INSTALL_BINDIR:PATH=bin

//Read-only architecture-independent data (DATAROOTDIR)
CMAKE_INSTALL_DATADIR:PATH=
[... 425 lines cut ...]
Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "CompilerIdCXX.vcxproj"

The CXX compiler identification is MSVC, found in "F:/simconnect-monitor/blockfactory/build/CMakeFiles/3.22.3/CompilerIdCXX/CompilerIdCXX.exe"

Detecting CXX compiler ABI info compiled with the following output:
Change Dir: F:/simconnect-monitor/blockfactory/build/CMakeFiles/CMakeTmp

Run Build Command(s):C:/Program Files (x86)/Microsoft Visual Studio/2017/BuildTools/MSBuild/15.0/Bin/MSBuild.exe cmTC_c2cb3.vcxproj /p:Configuration=Debug /p:Platform=Win32 /p:VisualStudioVersion=15.0 /v:m && Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 for .NET Framework

Copyright (C) Microsoft Corporation. All rights reserved.

  Microsoft (R) C/C++ Optimizing Compiler Version 19.16.27045 for x86

  Copyright (C) Microsoft Corporation.  All rights reserved.

  

  CMakeCXXCompilerABI.cpp

  cl /c /Zi /W3 /WX- /diagnostics:classic /Od /Ob0 /Oy- /D _MBCS /D WIN32 /D _WINDOWS /D "CMAKE_INTDIR=\"Debug\"" /Gm- /EHsc /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /GR /Fo"cmTC_c2cb3.dir\Debug\\" /Fd"cmTC_c2cb3.dir\Debug\vc141.pdb" /Gd /TP /analyze- /errorReport:queue "C:\Program Files\CMake\share\cmake-3.22\Modules\CMakeCXXCompilerABI.cpp"

  

  cmTC_c2cb3.vcxproj -> F:\simconnect-monitor\blockfactory\build\CMakeFiles\CMakeTmp\Debug\cmTC_c2cb3.exe
```

kevinwidjaja21, 2022-03-27:
So, it seems that somehow, `_matlab_64Build` return as false. I force it to always output 64-bit path by changing line 1658 to `set(_matlab_current_suffix ${_matlab_bin_suffix_64bits})` and it seems to resolve the issue.

But anotehr issue pop up when trying to run `cmake --build . --target install`
```shell
$ cmake --build . --target install
Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Auto build dll exports
  shlibpp.vcxproj -> F:\simconnect-monitor\blockfactory\build\bin\Debug\shlibppd.dll
  Auto build dll exports
  Core.vcxproj -> F:\simconnect-monitor\blockfactory\build\bin\Debug\BlockFactoryCored.dll
  Auto build dll exports
     Creating library F:/simconnect-monitor/blockfactory/build/lib/Debug/mxppd.lib and object F:/simconnect-monitor/bl
  ockfactory/build/lib/Debug/mxppd.exp
MxArray.obj : error LNK2019: unresolved external symbol _mxFree_800 referenced in function "public: bool __thiscall mx
pp::MxArray::asString(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > &)" (?as
String@MxArray@mxpp@@QAE_NAAV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z) [F:\simconnect-monitor\b 
lockfactory\build\deps\mxpp\mxpp.vcxproj]
MxArray.obj : error LNK2019: unresolved external symbol _mxGetNumberOfDimensions_800 referenced in function "public: _ 
_thiscall mxpp::MxArray::MxArray(struct mxArray_tag const *,bool)" (??0MxArray@mxpp@@QAE@PBUmxArray_tag@@_N@Z) [F:\sim 
connect-monitor\blockfactory\build\deps\mxpp\mxpp.vcxproj]
MxArray.obj : error LNK2019: unresolved external symbol _mxGetDimensions_800 referenced in function "public: __thiscal 
l mxpp::MxArray::MxArray(struct mxArray_tag const *,bool)" (??0MxArray@mxpp@@QAE@PBUmxArray_tag@@_N@Z) [F:\simconnect- 
monitor\blockfactory\build\deps\mxpp\mxpp.vcxproj]
MxArray.obj : error LNK2019: unresolved external symbol _mxGetM_800 referenced in function "public: __thiscall mxpp::M 
xArray::MxArray(struct mxArray_tag const *,bool)" (??0MxArray@mxpp@@QAE@PBUmxArray_tag@@_N@Z) [F:\simconnect-monitor\b 
lockfactory\build\deps\mxpp\mxpp.vcxproj]
MxArray.obj : error LNK2019: unresolved external symbol _mxGetFieldNameByNumber_800 referenced in function "public: bo 
ol __thiscall mxpp::MxArray::asMxStruct(class std::unordered_map<class std::basic_string<char,struct std::char_traits< 
char>,class std::allocator<char> >,class std::shared_ptr<class mxpp::MxArray>,struct std::hash<class std::basic_string 
<char,struct std::char_traits<char>,class std::allocator<char> > >,struct std::equal_to<class std::basic_string<char,s 
truct std::char_traits<char>,class std::allocator<char> > >,class std::allocator<struct std::pair<class std::basic_str 
ing<char,struct std::char_traits<char>,class std::allocator<char> > const ,class std::shared_ptr<class mxpp::MxArray>  
> > > &)" (?asMxStruct@MxArray@mxpp@@QAE_NAAV?$unordered_map@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@ 
std@@V?$shared_ptr@VMxArray@mxpp@@@2@U?$hash@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@U?$equal 
_to@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@V?$allocator@U?$pair@$$CBV?$basic_string@DU?$char 
_traits@D@std@@V?$allocator@D@2@@std@@V?$shared_ptr@VMxArray@mxpp@@@2@@std@@@2@@std@@@Z) [F:\simconnect-monitor\blockf 
actory\build\deps\mxpp\mxpp.vcxproj]
MxArray.obj : error LNK2019: unresolved external symbol _mxGetFieldByNumber_800 referenced in function "public: bool _ 
_thiscall mxpp::MxArray::asMxStruct(class std::unordered_map<class std::basic_string<char,struct std::char_traits<char 
>,class std::allocator<char> >,class std::shared_ptr<class mxpp::MxArray>,struct std::hash<class std::basic_string<cha 
r,struct std::char_traits<char>,class std::allocator<char> > >,struct std::equal_to<class std::basic_string<char,struc 
t std::char_traits<char>,class std::allocator<char> > >,class std::allocator<struct std::pair<class std::basic_string< 
char,struct std::char_traits<char>,class std::allocator<char> > const ,class std::shared_ptr<class mxpp::MxArray> > >  
> &)" (?asMxStruct@MxArray@mxpp@@QAE_NAAV?$unordered_map@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@ 
@V?$shared_ptr@VMxArray@mxpp@@@2@U?$hash@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@U?$equal_to@ 
V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@V?$allocator@U?$pair@$$CBV?$basic_string@DU?$char_tra 
its@D@std@@V?$allocator@D@2@@std@@V?$shared_ptr@VMxArray@mxpp@@@2@@std@@@2@@std@@@Z) [F:\simconnect-monitor\blockfacto 
ry\build\deps\mxpp\mxpp.vcxproj]
MxArray.obj : error LNK2019: unresolved external symbol _mxGetCell_800 referenced in function "public: bool __thiscall 
 mxpp::MxArray::asMxCell(class std::vector<class std::shared_ptr<class mxpp::MxArray>,class std::allocator<class std:: 
shared_ptr<class mxpp::MxArray> > > &)" (?asMxCell@MxArray@mxpp@@QAE_NAAV?$vector@V?$shared_ptr@VMxArray@mxpp@@@std@@V 
?$allocator@V?$shared_ptr@VMxArray@mxpp@@@std@@@2@@std@@@Z) [F:\simconnect-monitor\blockfactory\build\deps\mxpp\mxpp.v 
cxproj]
MxArray.obj : error LNK2019: unresolved external symbol _mxGetClassID_800 referenced in function "public: __thiscall m 
xpp::MxArray::MxArray(struct mxArray_tag const *,bool)" (??0MxArray@mxpp@@QAE@PBUmxArray_tag@@_N@Z) [F:\simconnect-mon 
itor\blockfactory\build\deps\mxpp\mxpp.vcxproj]
[... cut ...]
get machine type 'x86' [F:\simconnect-monitor\blockfactory\build\deps\mxpp\mxpp.vcxproj]
F:\simconnect-monitor\blockfactory\build\bin\Debug\mxppd.dll : fatal error LNK1120: 20 unresolved externals [F:\simcon    
nect-monitor\blockfactory\build\deps\mxpp\mxpp.vcxproj]
  Auto build dll exports
  SimulinkCoder.vcxproj -> F:\simconnect-monitor\blockfactory\build\bin\Debug\BlockFactorySimulinkCoderd.dll
  blockfactory-exists.vcxproj -> F:\simconnect-monitor\blockfactory\build\bin\Debug\blockfactory-exists.exe
```
Not sure what is wrong now.

traversaro, 2022-03-27:
The problem is more clear for this error: your target/platform is x86 (i.e. 32-bit) while the MATLAB libraries are for 64-bit.

kevinwidjaja21, 2022-03-27:
Ok, that is weird. I am on 64-bit windows 10, with 64-bit MatLab and CMake. The only thing that is 32-bit is visual studio but 64-bit only just come this year. Is that the issue?

traversaro, 2022-03-27:
I may be wrong, but on VS2017 CMake still defaults to generate 32-bit projects by default (they switched to 64-bit by default in VS2019). You probably just need to pass -A x64 option to CMake first configuration to make sure you generate a 64-bit project.