robotology/blockfactory #69: Extension to newer versions of Matlab - 2020b:

mlaudu, 2023-02-06:
I was able to successfully compile the code in Windows 10. I need the toolbox to generate Simulink block for Real-time SpeedGoat machine which only runs on Matlab 2020b. However the example problem in Model.mdl will not work when I try it on this version of Matlab. Weird (unreadable) error messages were thrown. Then I installed Matlab 2018b and the example problem (Model.mdl) worked perfectly! Is there any strategy that I can use to make the code to work in Matlab 2020b? I will greatly appreciate any advise from experienced users.
Thanks.

traversaro, 2023-05-17:
Sorry @mlaudu, I had missed this message. What is the error you get with Matlab 2020b? In our experiments we have libraries that work fine on 2020b .

traversaro, 2023-05-17:
Not sure if this is related to https://github.com/robotology/blockfactory/issues/70 .

traversaro, 2024-09-23:
@mlaudu we will close given the lack of feedback, if you still have problems feel free to open a new issue, thank!

closed 2024-09-23 by traversaro