robotology/blockfactory #70: Errors running the Model.mdl of the example folder

davidegorbani, 2023-05-17:
I was trying to run the `Model.mdl` file of the `example` folder but I got the following errors in Simulink:

`Error:Error reported by S-function 'BlockFactory' in 'Model/w// mask':
Failed to create factory
Factory error (2): ExampleToolbox: cannot open shared object file: No such file or directory
Failed to get factory object (className=SignalMath,libName=ExampleToolbox)
Failed to get the factory
`

`Error:Error reported by S-function 'BlockFactory' in 'Model/w//o mask':
Failed to create factory
Factory error (2): ExampleToolbox: cannot open shared object file: No such file or directory
Failed to get factory object (className=SignalMath,libName=ExampleToolbox)
Failed to get the factory
`

I am using the Matlab version R2023a on Ubuntu 22.04; can this problem be related to my version of Matlab or Ubuntu or is it related to something else?

davidegorbani, 2023-05-17:
The output is:
`ERROR: Failed to get factory object (blockName=SignalMath,pluginName=ExampleToolbox`

davidegorbani, 2023-05-17:
I set it to `/home/dgorbani/software/robotology-superbuild/build` in the `./bashrc` but it is still not working and the output of `blockfactory-exists ExampleToolbox SignalMath` is still 
`ERROR: Failed to get factory object (blockName=SignalMath,pluginName=ExampleToolbox)`

davidegorbani, 2023-05-17:
I noticed that if I set `export BLOCKFACTORY_PLUGIN_PATH=/home/dgorbani/software/robotology-superbuild/build` then the output of `blockfactory-exists WBToolbox ForwardKinematics` is `ERROR: Failed to get factory object (blockName=ForwardKinematics,pluginName=WBToolbox)`; while if I set `export BLOCKFACTORY_PLUGIN_PATH=/home/dgorbani/software/robotology-superbuild/build/install/lib/blockfactory` the output of `blockfactory-exists WBToolbox ForwardKinematics` is `SUCCESS: Block "ForwardKinematics" found and loaded from plugin "WBToolbox"`. I don't know if this can help.

traversaro, 2023-05-17:
> I noticed that if I set `export BLOCKFACTORY_PLUGIN_PATH=/home/dgorbani/software/robotology-superbuild/build` then the output of `blockfactory-exists WBToolbox ForwardKinematics` is `ERROR: Failed [...]

I guess this is expected. `BLOCKFACTORY_PLUGIN_PATH` should point to the directory in which the `.so` to load is found, and in the case of `WBToolbox.so` the installed version is in `/home/dgorbani/software/robotology-superbuild/build/install/lib/blockfactory`, but the one in the build directory is not in `/home/dgorbani/software/robotology-superbuild/build` , it is probably in `/home/dgorbani/software/robotology-superbuild/build/src/WBToolbox`  or some subdirectory. However, to debug now probably it is easier if we concentrate on the `ExampleToolbox` case.

traversaro, 2023-05-17:
> I set it to `/home/dgorbani/software/robotology-superbuild/build` in the `./bashrc` but it is still not working and the output of `blockfactory-exists ExampleToolbox SignalMath` is still: `ERROR: Failed to get factory object (blockName=SignalMath,pluginName=ExampleToolbox)`

I think I got the misunderstanding. The `build` that needs to be added to the `BLOCKFACTORY_PLUGIN_PATH` env variable is the build directory of `blockfactory/example` one, (that I guess it is in `/home/dgorbani/software/robotology-superbuild/src/BlockFactory/example/build`), not the build of the superbuild.

davidegorbani, 2023-05-17:
Ok, thank you! The right folder is `/home/dgorbani/software/robotology-superbuild/build/src/BlockFactory/lib` and I also found out that in the `CMakelists.txt` of the BlockFactory folder is not present a `add_subdirectory(example)`. I added this line and when I recompiled the robotology superbuild I got the following error:

```
[ 86%] Built target ExampleToolbox
[ 90%] Building CXX object example/matlab/CMakeFiles/AutogenerationExample.dir/AutogenerationExample_grt_rtw/AutogenerationExample.cpp.o
In file included from /home/dgorbani/software/robotology-superbuild/src/BlockFactory/example/matlab/AutogenerationExample_grt_rtw/AutogenerationExample.h:27,
                 from /home/dgorbani/software/robotology-superbuild/src/BlockFactory/example/matlab/AutogenerationExample_grt_rtw/AutogenerationExample.cpp:21:
/home/dgorbani/software/robotology-superbuild/src/BlockFactory/example/matlab/AutogenerationExample_grt_rtw/rtwtypes.h:23:10: fatal error: tmwtypes.h: No such file or directory
   23 | #include "tmwtypes.h"
      |          ^~~~~~~~~~~~
compilation terminated.
make[5]: *** [example/matlab/CMakeFiles/AutogenerationExample.dir/build.make:76: example/matlab/CMakeFiles/AutogenerationExample.dir/AutogenerationExample_grt_rtw/AutogenerationExample.cpp.o] Error 1
make[4]: *** [CMakeFiles/Makefile2:533: example/matlab/CMakeFiles/AutogenerationExample.dir/all] Error 2
make[3]: *** [Makefile:136: all] Error 2
make[2]: *** [CMakeFiles/BlockFactory.dir/build.make:85: src/BlockFactory/CMakeFiles/YCMStamp/BlockFactory-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:1752: CMakeFiles/BlockFactory.dir/all] Error 2
make: *** [Makefile:101: all] Error 2
```

I commented out the part of the `CMakeLists.txt` of the `example` folder in which it autogenerates code and now `Model.mdl` is working fine.

traversaro, 2023-05-17:
> I also found out that in the `CMakelists.txt` of the BlockFactory folder is not present a `add_subdirectory(example)`.

No, the `example` is mean to be compile on its own as a project separate by BlockFactory, you are supposed to create a separate new build directory in example, did you tried to follow the steps in https://github.com/robotology/blockfactory/tree/master/example#build-the-example ?

davidegorbani, 2023-05-18:
Closing this is issue since this problem is solved

closed 2023-05-18 by davidegorbani

rcone11, 2024-05-06:
I encountered the same problem and followed the instructions on the webpage export BLOCKFACTORY_PLUGIN_PATH=/home/han/blockfactory/example/build. “SUCCESS: Block ‘SignalMath’ found and loaded from plugin ‘ExampleToolbox’” appeared, but I still received an error in Simulink: Error reported by S-function ‘BlockFactory’ in ‘Model/w// mask’: Failed to create factory. Factory error (2): ExampleToolbox: cannot open shared object file: No such file or directory. Failed to get factory object (className=SignalMath, libName=ExampleToolbox). Failed to get the factory.