robotology/blockfactory #71: Compilation Failed in Ubuntu with MATLAB2023b

EhsanRanjbari, 2024-01-04:
The compilation of robotology-superbuild failed with this error:
```bash
In file included from /home/Ehsan/robotology-superbuild/src/BlockFactory/deps/sharedlibpp/src/SharedLibraryFactory.cpp:9:
/home/Ehsan/robotology-superbuild/src/BlockFactory/deps/sharedlibpp/src/shlibpp/SharedLibraryFactory.h:37:10: warning: elaborated-type-specifier for a scoped enum must not use the ‘class’ keyword
   37 |     enum class Status : std::uint32_t
      |     ~~~~ ^~~~~
      |          -----
/home/Ehsan/robotology-superbuild/src/BlockFactory/deps/sharedlibpp/src/shlibpp/SharedLibraryFactory.h:37:16: error: use of enum ‘Status’ without previous declaration
   37 |     enum class Status : std::uint32_t
      |                ^~~~~~
/home/Ehsan/robotology-superbuild/src/BlockFactory/deps/sharedlibpp/src/shlibpp/SharedLibraryFactory.h:37:30: error: ‘uint32_t’ is not a member of ‘std’; did you mean ‘wint_t’?
   37 |     enum class Status : std::uint32_t
      |                              ^~~~~~~~
      |                              wint_t
/home/Ehsan/robotology-superbuild/src/BlockFactory/deps/sharedlibpp/src/shlibpp/SharedLibraryFactory.h:38:5: error: default member initializer for unnamed bit-field
   38 |     {
      |     ^
/home/Ehsan/robotology-superbuild/src/BlockFactory/deps/sharedlibpp/src/shlibpp/SharedLibraryFactory.h:169:5: error: ‘Status’ does not name a type
  169 |     Status getStatus() const;
      |     ^~~~~~
/home/Ehsan/robotology-superbuild/src/BlockFactory/deps/sharedlibpp/src/SharedLibraryFactory.cpp:52:27: error: ‘Status’ in ‘class shlibpp::SharedLibraryFactory’ does not name a type
   52 |     SharedLibraryFactory::Status status;
      |                           ^~~~~~
/home/Ehsan/robotology-superbuild/src/BlockFactory/deps/sharedlibpp/src/SharedLibraryFactory.cpp: In constructor ‘shlibpp::SharedLibraryFactory::Private::Private(int32_t, int32_t, int32_t, const char*)’:
/home/Ehsan/robotology-superbuild/src/BlockFactory/deps/sharedlibpp/src/SharedLibraryFactory.cpp:76:9: error: class ‘shlibpp::SharedLibraryFactory::Private’ does not have any field named ‘status’
   76 |         status(Status::None),
      |         ^~~~~~
/home/Ehsan/robotology-superbuild/src/BlockFactory/deps/sharedlibpp/src/SharedLibraryFactory.cpp:76:16: error: ‘Status’ has not been declared
   76 |         status(Status::None),
      |                ^~~~~~
/home/Ehsan/robotology-superbuild/src/BlockFactory/deps/sharedlibpp/src/SharedLibraryFactory.cpp: In member function ‘bool shlibpp::SharedLibraryFactory::Private::open(const char*)’:
/home/Ehsan/robotology-superbuild/src/BlockFactory/deps/sharedlibpp/src/SharedLibraryFactory.cpp:131:5: error: ‘status’ was not declared in this scope; did you mean ‘statx’?
  131 |     status = Status::None;
      |     ^~~~~~
      |     statx
/home/Ehsan/robotology-superbuild/src/BlockFactory/deps/sharedlibpp/src/SharedLibraryFactory.cpp:131:14: error: ‘Status’ has not been declared
  131 |     status = Status::None;
      |              ^~~~~~
/home/Ehsan/robotology-superbuild/src/BlockFactory/deps/sharedlibpp/src/SharedLibraryFactory.cpp:145:22: error: ‘Status’ has not been declared
  145 |             status = Status::LibraryNotFound;
      |                      ^~~~~~
/home/Ehsan/robotology-superbuild/src/BlockFactory/deps/sharedlibpp/src/SharedLibraryFactory.cpp:147:22: error: ‘Status’ has not been declared
  147 |             status = Status::LibraryNotLoaded;
      |                      ^~~~~~
/home/Ehsan/robotology-superbuild/src/BlockFactory/deps/sharedlibpp/src/SharedLibraryFactory.cpp:154:18: error: ‘Status’ has not been declared
  154 |         status = Status::FactoryNotFound;
      |                  ^~~~~~
/home/Ehsan/robotology-superbuild/src/BlockFactory/deps/sharedlibpp/src/SharedLibraryFactory.cpp:160:18: error: ‘Status’ has not been declared
  160 |         status = Status::FactoryNotFunctional;
      |                  ^~~~~~
/home/Ehsan/robotology-superbuild/src/BlockFactory/deps/sharedlibpp/src/SharedLibraryFactory.cpp:164:14: error: ‘Status’ has not been declared
  164 |     status = Status::OK;
      |              ^~~~~~
/home/Ehsan/robotology-superbuild/src/BlockFactory/deps/sharedlibpp/src/SharedLibraryFactory.cpp: At global scope:
/home/Ehsan/robotology-superbuild/src/BlockFactory/deps/sharedlibpp/src/SharedLibraryFactory.cpp:316:32: error: ‘Status’ in ‘class shlibpp::SharedLibraryFactory’ does not name a type
  316 | shlibpp::SharedLibraryFactory::Status shlibpp::SharedLibraryFactory::getStatus() const
      |                                ^~~~~~
make[2]: *** [deps/sharedlibpp/src/CMakeFiles/shlibpp.dir/build.make:104: deps/sharedlibpp/src/CMakeFiles/shlibpp.dir/SharedLibraryFactory.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:224: deps/sharedlibpp/src/CMakeFiles/shlibpp.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

traversaro, 2024-01-04:
Interesting, the error does not see ln at a glance related to the MATLAB version. Can you provide more info on:
* Distro/os version used
* PKG manager used to install dependencies (if conda, report also conda list)
* anything that may be strange in your setup

EhsanRanjbari, 2024-01-08:
> * Distro/os version used

Ubuntu 22.04.3 LTS.

> * PKG manager used to install dependencies (if conda, report also conda list)

apt (followed https://github.com/robotology/robotology-superbuild?tab=readme-ov-file#source-installation)

> * anything that may be strange in your setup

only this error: https://github.com/robotology/yarp/issues/2885#issuecomment-1876782445

EhsanRanjbari, 2024-01-09:
> > * Distro/os version used
> 
> Ubuntu 22.04.3 LTS.
> 
> > * PKG manager used to install dependencies (if conda, report also conda list)
> 
> apt (followed https://github.com/robotology/robotology-superbuild?tab=readme-ov-file#source-installation)
> 
> > * anything that may be strange in your setup
> 
> only this error: [robotology/yarp#2885 (comment)](https://github.com/robotology/yarp/issues/2885#issuecomment-1876782445)

The issue https://github.com/robotology/yarp/issues/2885#issuecomment-1876782445 seems to be resolved. But still getting the BlockFactory error mentioned in the first comment.

traversaro, 2024-01-09:
Interesting, there is indeed a bug, but I do not know how nobody ever noticed this. Can you provide the `build/src/blockfactory/CMakeCache.txt` file? Perhaps are you using a compiler different from the standard one provided in Ubuntu 22.04?

traversaro, 2024-01-09:
@EhsanRanjbari can you test if https://github.com/robotology/blockfactory/pull/72 fixes your problem?

EhsanRanjbari, 2024-01-10:
Here is `build/src/blockfactory/CMakeCache.txt`: [CMakeCache.txt](https://github.com/robotology/blockfactory/files/13886935/CMakeCache.txt).

And #72 fixed the problem. Thanks. I will close the issue shortly.

traversaro, 2024-01-10:
Perfect, I releases 0.8.4 with this fix: https://github.com/robotology/blockfactory/releases/tag/v0.8.4 .

closed 2024-01-10 by traversaro