robotology/blockfactory #77: s-function

rcone11, 2024-04-29:
I am trying to install block factory for the first time. I want to run the example in the blockfactory folder first. When I tried to run it in Simulink, I got an error saying it couldn’t find blockfactory. I also tried to mex blockfactory.cpp, but I couldn’t compile it successfully. I want to know if I missed any steps. What should I do to make Simulink find blockfactory?

traversaro, 2024-04-29:
Please report the exact error, thanks.

rcone11, 2024-04-29:
I tried to mex blockfactory.cpp in matlab and got the error message: /usr/bin/ld: /tmp/mex_35266495627694_401685/BlockFactory.o: in function catchLogMessages(bool, SimStruct_tag*)': BlockFactory.cpp:(.text+0x123a): undefined reference to blockfactory::core::Log::getSingleton()’
/usr/bin/ld: BlockFactory.cpp:(.text+0x124c): undefined reference to `blockfactory::core::Log::getWarnings[abi:cxx11](chrome-extension://bpelnogcookhocnaokfpoeinibimbeff/index.html)
const’, this is just a small part of the error report.what should I do?thank you

rcone11, 2024-04-29:
> 请报告确切的错误，谢谢。

I opened the model in the example in simulink, and after running it, I encountered the following error: Error in S-function ‘Model/w// mask’: S-Function ‘BlockFactory’ does not exist Component: Simulink | Category: Model error Error in S-function ‘Model/w//o mask’: S-Function ‘BlockFactory’ does not exist Component: Simulink | Category: Model error. I have been trying to solve this problem, and I was thinking that if I successfully compile block factory.cpp in Matlab, this error would not exist. But I have been unsuccessful, maybe my approach is wrong?

traversaro, 2024-04-29:
The S-Function needs to be compiled via CMake, see https://robotology.github.io/blockfactory/mkdocs/install/#installation .

rcone11, 2024-04-29:
> The S-Function needs to be compiled via CMake, see https://robotology.github.io/blockfactory/mkdocs/install/#installation .

On this webpage, I have completed everything as instructed on Ubuntu 20.04.6, except for the last step regarding export BLOCKFACTORY_PLUGIN_PATH=/installfolder1/subfolder1/blockfactory:/installfolder2/subfolder2/blockfactory/:<...>. because I am not very clear on which files I need to export, can you please help me with this, sir?thank you

traversaro, 2024-04-29:
That is an environment variable, see https://www.hostinger.com/tutorials/linux-environment-variables for a primer on that. You only need to set it once you have compiled your first blockfactory block, see https://robotology.github.io/blockfactory/mkdocs/create_new_library/ for an example on how to do that.

rcone11, 2024-04-29:
> 这是一个环境变量，有关这方面的入门知识，请参见 https://www.hostinger.com/tutorials/linux-environment-variables。您只需要在编译第一个 blockfactory 块后进行设置，有关如何执行此操作的示例，请参阅 https://robotology.github.io/blockfactory/mkdocs/create_new_library/。

So,sir  do I not need to try mex blockfactory in Matlab? Just need to set export BLOCKFACTORY_PLUGIN_PATH=/installfolder1/subfolder1/blockfactory:/installfolder2/subfolder2/blockfactory/:<…> right, so that the example can run fine right? and on (https://robotology.github.io/blockfactory/mkdocs/create_new_library/) “You should now find in the ./build directories a new library file, which depending on your OS is: libExampleToolbox.so on Linux.” After I executed this, I did not find the libExampleToolbox.so, why is that? thank you

traversaro, 2024-04-29:
Did you followed the instructions in the "Compiling instructions" in https://robotology.github.io/blockfactory/mkdocs/create_new_library/ ?

rcone11, 2024-04-29:
> Did you followed the instructions in the "Compiling instructions" in https://robotology.github.io/blockfactory/mkdocs/create_new_library/ ?

yes,I followed this command

traversaro, 2024-04-29:
Can you report the output of those commands?

rcone11, 2024-04-29:
> Can you report the output of those commands?

after the last command:Found BlockFactory 0.8.5 (Components: Core Simulink SimulinkCoder)
-- Found BlockFactory 0.8.5 (Components: Core Simulink SimulinkCoder)
-- Found BlockFactory 0.8.5 (Components: Core Simulink SimulinkCoder)
-- Found autogenerated sources for AutogenerationExample model.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/han/blockfactory/example/build

traversaro, 2024-04-29:
> after the last command:Found BlockFactory 0.8.5 (Components: Core Simulink SimulinkCoder) -- Found BlockFactory 0.8.5 (Components: Core Simulink SimulinkCoder) -- Found BlockFactory 0.8.5 [...]

It seems that you did not executed the command `cmake --build .`, unless I am missing something.

rcone11, 2024-04-29:
> It seems that you did not executed the command `cmake --build .`, unless I am missing something.

When I execute cmake build ., an error occurs:
[ 42%] Linking CXX executable main
/usr/bin/ld: /home/han/miniforge3/lib/libBlockFactorySimulinkCoder.so.0.8.5: undefined reference to `std::__throw_bad_array_new_length()@GLIBCXX_3.4.29’
collect2: error: ld returned 1
make[2]: *** [CMakeFiles/main.dir/build.make:88：main] Error 1
make[1]: *** [CMakeFiles/Makefile2:98：CMakeFiles/main.dir/all] Error 2
make: *** [Makefile:84：all] Error 2

Then I changed the command to cmake -Dcmake_CXX_STANDARD=14 --build ., the output is:
Found BlockFactory 0.8.5 (Components: Core Simulink SimulinkCoder)
– Found BlockFactory 0.8.5 (Components: Core Simulink SimulinkCoder)
– Found BlockFactory 0.8.5 (Components: Core Simulink SimulinkCoder)
– Found autogenerated sources for AutogenerationExample model.
– Configuring done
– Generating done
– Build files have been written to: /home/han/blockfactory/example/build。It's wrong?

traversaro, 2024-04-29:
Please always report the full errors you are encountering, otherwise it is really difficult to help you. : )

In the command `cmake -Dcmake_CXX_STANDARD=14 --build .` the `--build` part is ignored as you are passing a `-D` option, so basically you are not building anymore the example.

It seems that you are building inside a conda environement, can you please report the content of that environment with the output `conda list` ? My intuition is that you are trying to use libraries from that conda environment, but without installing the compilers for conda, but using the compilers from the underlying distribution, but that is not possible, as the libraries from conda require a newer C++ standard library, and hence the error you are seeing.

traversaro, 2024-04-29:
Can you report the output of the `conda list` command?

rcone11, 2024-04-29:
> conda list

(base) han@han-System-Product-Name:~$ conda list
# packages in environment at /home/han/miniforge3:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
ace                       7.1.3                h59595ed_2    conda-forge
alsa-lib                  1.2.11               hd590300_1    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
aom                       3.8.2                h59595ed_0    conda-forge
archspec                  0.2.3              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                h8343317_3    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
blockfactory              0.8.5              h59595ed_120    robotology
boltons                   24.0.0             pyhd8ed1ab_0    conda-forge
brotli-python             1.1.0           py310hc6cd4ac_1    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
c-ares                    1.28.1               hd590300_0    conda-forge
ca-certificates           2024.2.2             hbcca054_0    conda-forge
cairo                     1.18.0               h3faef2a_0    conda-forge
casadi                    3.6.5           py310h60ac289_0    conda-forge
casadi-matlab-bindings    3.6.3.1            h59595ed_120    robotology
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py310h2fee648_0    conda-forge
charset-normalizer        3.3.2              pyhd8ed1ab_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
conda                     24.3.0          py310hff52083_0    conda-forge
conda-libmamba-solver     24.1.0             pyhd8ed1ab_0    conda-forge
conda-package-handling    2.2.0              pyh38be061_0    conda-forge
conda-package-streaming   0.9.0              pyhd8ed1ab_0    conda-forge
cython                    0.29.37         py310hc6cd4ac_0    conda-forge
dav1d                     1.2.1                hd590300_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
distro                    1.9.0              pyhd8ed1ab_0    conda-forge
eigen                     3.4.0                h00ab1b0_0    conda-forge
expat                     2.6.2                h59595ed_0    conda-forge
ffmpeg                    6.1.1           gpl_hee4b679_108    conda-forge
fmt                       10.2.1               h00ab1b0_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_1    conda-forge
fontconfig                2.14.2               h14ed4e7_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeglut                  3.2.2                hac7e632_2    conda-forge
freetype                  2.12.1               h267a509_2    conda-forge
fribidi                   1.0.10               h36c2ea0_0    conda-forge
gettext                   0.22.5               h59595ed_2    conda-forge
gettext-tools             0.22.5               h59595ed_2    conda-forge
glfw                      3.4                  hd590300_0    conda-forge
glib                      2.80.0               hf2295e7_6    conda-forge
glib-tools                2.80.0               hde27a5a_6    conda-forge
gmp                       6.3.0                h59595ed_1    conda-forge
gnutls                    3.7.9                hb077bed_0    conda-forge
graphite2                 1.3.13            h59595ed_1003    conda-forge
gsl                       2.7                  he838d99_0    conda-forge
gst-plugins-base          1.24.1               hfa15dee_1    conda-forge
gstreamer                 1.24.1               h98fc4e7_1    conda-forge
harfbuzz                  8.3.0                h3d44ed6_0    conda-forge
hdf5                      1.14.3          nompi_h4f84152_100    conda-forge
icu                       73.2                 h59595ed_0    conda-forge
icub-firmware-shared      1.38.0             h59595ed_120    robotology
icub-main                 2.5.0           py310hd29ae9e_120    robotology
icub-models               2.5.0           py310hc6cd4ac_0    conda-forge
idna                      3.6                pyhd8ed1ab_0    conda-forge
idyntree                  11.0.0          py310h3478c64_0    conda-forge
idyntree-matlab-bindings  10.3.0             h9f71af4_120    robotology
[... 184 lines cut ...]
xorg-fixesproto           5.0               h7f98852_1002    conda-forge
xorg-inputproto           2.3.2             h7f98852_1002    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libice               1.1.1                hd590300_0    conda-forge
xorg-libsm                1.2.4                h7391055_0    conda-forge
xorg-libx11               1.8.9                h8ee46fc_0    conda-forge
xorg-libxau               1.0.11               hd590300_0    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
[... cut ...]
xz                        5.2.6                h166bdaf_0    conda-forge
yaml-cpp                  0.8.0                h59595ed_0    conda-forge
yarp-matlab-bindings      3.9.1              h59595ed_120    robotology
ycm-cmake-modules         0.16.2               h59595ed_0    conda-forge
zlib                      1.2.13               hd590300_5    conda-forge
zstandard                 0.22.0          py310h1275a96_0    conda-forge
zstd                      1.5.5                hfc55251_0    conda-forge
(base) han@han-System-Product-Name:~$

traversaro, 2024-04-29:
My theory is correct, there is no `compilers` package. Either you deactivate the conda environment when you build blockfactory and all related packages, or you also install in the environment the packages required for C++ development, i.e. :
~~~
conda install cmake pkg-config make ninja compilers
~~~

rcone11, 2024-04-29:
> > My theory is correct, there is no  package. Either you deactivate the conda environment when you build blockfactory and all related packages, or you also install in the environment the packages required for C++ development, i.e. :`compilers`
> > ```
> > conda install cmake pkg-config make ninja compilers
> > ```
> 
> I will follow your instructions and appreciate your patient guidance .thank you very much

I am sorry to bother you again. When my terminal was executing cmake --build . --target install, an error occurred in this last step:

han@han-System-Product-Name:~/blockfactory/build$ cmake --build . --target install
make[1]: Entering directory “/home/han/blockfactory/build”
make[2]: Entering directory “/home/han/blockfactory/build”
make[2]: Leaving directory “/home/han/blockfactory/build”
[ 25%] Built target shlibpp
make[2]: Entering directory “/home/han/blockfactory/build”
make[2]: Leaving directory “/home/han/blockfactory/build”
[ 75%] Built target Core
make[2]: Entering directory “/home/han/blockfactory/build”
make[2]: Leaving directory “/home/han/blockfactory/build”
[ 87%] Built target SimulinkCoder
make[2]: Entering directory “/home/han/blockfactory/build”
make[2]: Leaving directory “/home/han/blockfactory/build”
[100%] Built target blockfactory-exists
make[1]: Leaving directory “/home/han/blockfactory/build”
Install the project…
– Install configuration: “Release”
– Installing: /home/han/blockfactory/build/lib/cmake/shlibpp/shlibppConfigVersion.cmake
– Up-to-date: /home/han/blockfactory/build/lib/cmake/shlibpp/shlibppConfig.cmake
– Old export file “/home/han/blockfactory/build/lib/cmake/shlibpp/shlibppTargets.cmake” will be replaced. Removing files [/home/han/blockfactory/build/lib/cmake/shlibpp/shlibppTargets-release.cmake].
– Installing: /home/han/blockfactory/build/lib/cmake/shlibpp/shlibppTargets.cmake
– Installing: /home/han/blockfactory/build/lib/cmake/shlibpp/shlibppTargets-release.cmake
CMake Error at deps/sharedlibpp/src/cmake_install.cmake:52 (file):
file INSTALL cannot find
“/home/han/blockfactory/build/lib/libshlibpp.so.0.0.1”: No such file or
directory.
Call Stack (most recent call first):
deps/sharedlibpp/cmake_install.cmake:69 (include)
deps/cmake_install.cmake:42 (include)
cmake_install.cmake:42 (include)

make: *** [Makefile:86: install] Error 1
han@han-System-Product-Name:~/blockfactory/build$

traversaro, 2024-04-29:
It seems that you already installed `blockfactory` in your conda environment, so I guess you do not need to re-compile it from source. I guess you just need to compile the example block.

rcone11, 2024-04-29:
> It seems that you already installed `blockfactory` in your conda environment, so I guess you do not need to re-compile it from source. I guess you just need to compile the example block.

Just now I deleted the blockfactory folder in the computer and followed the instructions step by step, then this error occurred. Does this mean that this error can be ignored?

rcone11, 2024-04-29:
> > > It seems that you already installed `blockfactory` in your conda environment, so I guess you do not need to re-compile it from source. I guess you just need to compile the example block.
> > 
> > 
> > Just now I deleted the blockfactory folder in the computer and followed the instructions step by step, then this error occurred. Does this mean that this error can be ignored?
> 
> Which error?

> > > It seems that you already installed `blockfactory` in your conda environment, so I guess you do not need to re-compile it from source. I guess you just need to compile the example block.
> > 
> > 
> > Just now I deleted the blockfactory folder in the computer and followed the instructions step by step, then this error occurred. Does this mean that this error can be ignored?
> 
> Which error?

I'm sorry, I didn't make myself clear. The error I mentioned were those in my previous reply.

rcone11, 2024-04-29:
> The error in [#77 (comment)](https://github.com/robotology/blockfactory/issues/77#issuecomment-2082452979) mention the folder `/home/han/blockfactory/build/`, so I am not sure what you mean with "I deleted the blockfactory folder", as the error refer to the `/home/han/blockfactory` folder. Anyhow, as I stated before if you already installed `blockfactory` via conda, you do not need to install it again from source (i.e. following the steps in https://robotology.github.io/blockfactory/mkdocs/install/).

OK,Thank you for your sincere answer and the great help you have given me.

traversaro, 2024-09-23:
Given the lack of reply I guess this was solved, @rcone11 if this is not the case, feel free to open a new issue, thanks!

closed 2024-09-23 by traversaro