robotology/blockfactory #79: Errors running the Model.mdl of the example folder

rcone11, 2024-05-06:
my BlockFactory.mexa64 file address is /home/han/miniforge3/mex/BlockFactory.mexa64.But my model.mdl file address is /home/han/blockfactory/example/matlab/model.mdl.To ensure that these two files are in the same folder，I copied the BlockFactory.mexa64 file to the /home/han/blockfactory/example/matlab/BlockFactory.mexa64.Then I run the model.mdl  here is the error displayed.
Error while obtaining sizes from MEX S-function 'BlockFactory in "odel/w// mask'Caused by:
MEX file "/home/hanblockfactory/example/matlab/BlockFactory.mexa64' invalid:libBleckFactoryCore.s9.0.8.5: cannot open shared object file : No such file or directoryComponent:Simwink Category:Model errorError while obtaining sizes from MEX S-function 'BlockFacto!y in 'Model/w//o mask'.Caused by:MEX file '/home/hablockfactory/example/matlab/BlockFactory.mexa64' invalid:libBleckFactoryCore.s9.0.8.5: cannot open shared object file : No such file or directory.

traversaro, 2024-05-06:
How did you compile the `/home/hablockfactory/example/matlab/BlockFactory.mexa64` file? The error seems to refer to a `libBleckFactoryCore`, not sure if the `Bleck`/`Block` typo is related.

rcone11, 2024-05-06:
> How did you compile the `/home/hablockfactory/example/matlab/BlockFactory.mexa64` file? The error seems to refer to a `libBleckFactoryCore`, not sure if the `Bleck`/`Block` typo is related.

sorry.There are some words spelled incorrectly， here is the error displayed.
Error while obtaining sizes from MEX S-function 'BlockFactory in "odel/w// mask'Caused by:
MEX file "/home/han/blockfactory/example/matlab/BlockFactory.mexa64' invalid:libBlockFactoryCore.so.0.8.5: cannot open shared object file : No such file or directory 
Component:Simulink  | Category:Model error
Error while obtaining sizes from MEX S-function 'BlockFacto!y in 'Model/w//o mask'.Caused by:
MEX file '/home/han/blockfactory/example/matlab/BlockFactory.mexa64' invalid:libBleckFactoryCore.so.0.8.5: cannot open shared object file : No such file or directory.

rcone11, 2024-05-06:
> The problem seems the same. Anyhow, the question is still valid:
> 
> > How did you compile the `/home/hablockfactory/example/matlab/BlockFactory.mexa64` file?
> 
> If you are using a conda environment, can you report the `conda list`?
I didn't compiled the  BlockFactory.mexa64 succeed 
conda list:
# packages in environment at /home/han/miniforge3:
#
# Name Version Build Channel
_libgcc_mutex 0.1 conda_forge conda-forge
_openmp_mutex 4.5 2_gnu conda-forge
ace 7.1.3 h59595ed_2 conda-forge
alsa-lib 1.2.11 hd590300_1 conda-forge
ampl-mp 3.1.0 h2cc385e_1006 conda-forge
aom 3.8.2 h59595ed_0 conda-forge
archspec 0.2.3 pyhd8ed1ab_0 conda-forge
assimp 5.3.1 h8343317_3 conda-forge
attr 2.5.1 h166bdaf_1 conda-forge
binutils 2.40 h4852527_0 conda-forge
binutils_impl_linux-64 2.40 hf600244_0 conda-forge
binutils_linux-64 2.40 hdade7a5_3 conda-forge
blockfactory 0.8.5 h59595ed_120 robotology
boltons 24.0.0 pyhd8ed1ab_0 conda-forge
brotli-python 1.1.0 py310hc6cd4ac_1 conda-forge
bzip2 1.0.8 hd590300_5 conda-forge
c-ares 1.28.1 hd590300_0 conda-forge
c-compiler 1.7.0 hd590300_0 conda-forge
ca-certificates 2024.2.2 hbcca054_0 conda-forge
cairo 1.18.0 h3faef2a_0 conda-forge
casadi 3.6.5 py310h60ac289_0 conda-forge
casadi-matlab-bindings 3.6.3.1 h59595ed_120 robotology
certifi 2024.2.2 pyhd8ed1ab_0 conda-forge
cffi 1.16.0 py310h2fee648_0 conda-forge
charset-normalizer 3.3.2 pyhd8ed1ab_0 conda-forge
cmake 3.29.2 hcfe8598_0 conda-forge
colorama 0.4.6 pyhd8ed1ab_0 conda-forge
compilers 1.7.0 ha770c72_0 conda-forge
conda 24.3.0 py310hff52083_0 conda-forge
conda-libmamba-solver 24.1.0 pyhd8ed1ab_0 conda-forge
conda-package-handling 2.2.0 pyh38be061_0 conda-forge
conda-package-streaming 0.9.0 pyhd8ed1ab_0 conda-forge
cxx-compiler 1.7.0 h00ab1b0_0 conda-forge
cython 0.29.37 py310hc6cd4ac_0 conda-forge
dav1d 1.2.1 hd590300_0 conda-forge
dbus 1.13.6 h5008d03_3 conda-forge
distro 1.9.0 pyhd8ed1ab_0 conda-forge
eigen 3.4.0 h00ab1b0_0 conda-forge
expat 2.6.2 h59595ed_0 conda-forge
ffmpeg 6.1.1 gpl_hee4b679_108 conda-forge
fmt 10.2.1 h00ab1b0_0 conda-forge
font-ttf-dejavu-sans-mono 2.37 hab24e00_0 conda-forge
font-ttf-inconsolata 3.000 h77eed37_0 conda-forge
font-ttf-source-code-pro 2.038 h77eed37_0 conda-forge
font-ttf-ubuntu 0.83 h77eed37_1 conda-forge
fontconfig 2.14.2 h14ed4e7_0 conda-forge
fonts-conda-ecosystem 1 0 conda-forge
fonts-conda-forge 1 0 conda-forge
fortran-compiler 1.7.0 heb67821_0 conda-forge
freeglut 3.2.2 hac7e632_2 conda-forge
freetype 2.12.1 h267a509_2 conda-forge
fribidi 1.0.10 h36c2ea0_0 conda-forge
gcc 12.3.0 h915e2ae_6 conda-forge
gcc_impl_linux-64 12.3.0 h1562d66_6 conda-forge
gcc_linux-64 12.3.0 h6477408_3 conda-forge
gettext 0.22.5 h59595ed_2 conda-forge
gettext-tools 0.22.5 h59595ed_2 conda-forge
gfortran 12.3.0 h915e2ae_6 conda-forge
gfortran_impl_linux-64 12.3.0 h6d6b2fb_6 conda-forge
gfortran_linux-64 12.3.0 h617cb40_3 conda-forge
[... 214 lines cut ...]
xkeyboard-config 2.41 hd590300_0 conda-forge
xorg-fixesproto 5.0 h7f98852_1002 conda-forge
xorg-inputproto 2.3.2 h7f98852_1002 conda-forge
xorg-kbproto 1.0.7 h7f98852_1002 conda-forge
xorg-libice 1.1.1 hd590300_0 conda-forge
xorg-libsm 1.2.4 h7391055_0 conda-forge
xorg-libx11 1.8.9 h8ee46fc_0 conda-forge
xorg-libxau 1.0.11 hd590300_0 conda-forge
xorg-libxdmcp 1.1.3 h7f98852_0 conda-forge
xorg-libxext 1.3.4 h0b41bf4_2 conda-forge
xorg-libxfixes 5.0.3 h7f98852_1004 conda-forge
xorg-libxi 1.7.10 h7f98852_0 conda-forge
xorg-libxinerama 1.1.5 h27087fc_0 conda-forge
xorg-libxrender 0.9.11 hd590300_0 conda-forge
xorg-renderproto 0.11.1 h7f98852_1002 conda-forge
xorg-xextproto 7.3.0 h0b41bf4_1003 conda-forge
xorg-xf86vidmodeproto 2.3.1 h7f98852_1002 conda-forge
xorg-xproto 7.0.31 h7f98852_1007 conda-forge
xz 5.2.6 h166bdaf_0 conda-forge
yaml-cpp 0.8.0 h59595ed_0 conda-forge
yarp-matlab-bindings 3.9.1 h59595ed_120 robotology
ycm-cmake-modules 0.16.2 h59595ed_0 conda-forge
zlib 1.2.13 hd590300_5 conda-forge
zstandard 0.22.0 py310h1275a96_0 conda-forge
zstd 1.5.5 hfc55251_0 conda-forge

traversaro, 2024-05-06:
Please read https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax on how to correctly format github comments (in particular use `~~~` blocks to contains the `conda list` output, thanks.

rcone11, 2024-05-07:
I'm sorry for my ignorance and I didn't  successfully compiled the /home/hablockfactory/example/matlab/BlockFactory.mexa64 file .

~~~
(base) han@han-System-Product-Name:~$ conda list
# packages in environment at /home/han/miniforge3:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
ace                        7.1.3                h59595ed_2    conda-forge
alsa-lib                  1.2.11               hd590300_1    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
aom                       3.8.2                h59595ed_0    conda-forge
archspec                  0.2.3              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                h8343317_3    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
binutils                  2.40                 h4852527_0    conda-forge
binutils_impl_linux-64    2.40                 hf600244_0    conda-forge
binutils_linux-64         2.40                 hdade7a5_3    conda-forge
blockfactory              0.8.5              h59595ed_120    robotology
boltons                   24.0.0             pyhd8ed1ab_0    conda-forge
brotli-python             1.1.0           py310hc6cd4ac_1    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
c-ares                    1.28.1               hd590300_0    conda-forge
c-compiler                1.7.0                hd590300_0    conda-forge
ca-certificates           2024.2.2             hbcca054_0    conda-forge
cairo                     1.18.0               h3faef2a_0    conda-forge
casadi                    3.6.5           py310h60ac289_0    conda-forge
casadi-matlab-bindings    3.6.3.1            h59595ed_120    robotology
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py310h2fee648_0    conda-forge
charset-normalizer        3.3.2              pyhd8ed1ab_0    conda-forge
cmake                     3.29.2               hcfe8598_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
compilers                 1.7.0                ha770c72_0    conda-forge
conda                     24.3.0          py310hff52083_0    conda-forge
conda-libmamba-solver     24.1.0             pyhd8ed1ab_0    conda-forge
conda-package-handling    2.2.0              pyh38be061_0    conda-forge
conda-package-streaming   0.9.0              pyhd8ed1ab_0    conda-forge
cxx-compiler              1.7.0                h00ab1b0_0    conda-forge
cython                    0.29.37         py310hc6cd4ac_0    conda-forge
dav1d                     1.2.1                hd590300_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
distro                    1.9.0              pyhd8ed1ab_0    conda-forge
eigen                     3.4.0                h00ab1b0_0    conda-forge
expat                     2.6.2                h59595ed_0    conda-forge
ffmpeg                    6.1.1           gpl_hee4b679_108    conda-forge
fmt                       10.2.1               h00ab1b0_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_1    conda-forge
fontconfig                2.14.2               h14ed4e7_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fortran-compiler          1.7.0                heb67821_0    conda-forge
freeglut                  3.2.2                hac7e632_2    conda-forge
freetype                  2.12.1               h267a509_2    conda-forge
fribidi                   1.0.10               h36c2ea0_0    conda-forge
gcc                       12.3.0               h915e2ae_6    conda-forge
gcc_impl_linux-64         12.3.0               h1562d66_6    conda-forge
gcc_linux-64              12.3.0               h6477408_3    conda-forge
gettext                   0.22.5               h59595ed_2    conda-forge
gettext-tools             0.22.5               h59595ed_2    conda-forge
gfortran                  12.3.0               h915e2ae_6    conda-forge
gfortran_impl_linux-64    12.3.0               h6d6b2fb_6    conda-forge
gfortran_linux-64         12.3.0               h617cb40_3    conda-forge
glfw                      3.4                  hd590300_0    conda-forge
glib                      2.80.0               hf2295e7_6    conda-forge
glib-tools                2.80.0               hde27a5a_6    conda-forge
[... 211 lines cut ...]
xkeyboard-config          2.41                 hd590300_0    conda-forge
xorg-fixesproto           5.0               h7f98852_1002    conda-forge
xorg-inputproto           2.3.2             h7f98852_1002    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libice               1.1.1                hd590300_0    conda-forge
xorg-libsm                1.2.4                h7391055_0    conda-forge
xorg-libx11               1.8.9                h8ee46fc_0    conda-forge
[... cut ...]
xz                        5.2.6                h166bdaf_0    conda-forge
yaml-cpp                  0.8.0                h59595ed_0    conda-forge
yarp-matlab-bindings      3.9.1              h59595ed_120    robotology
ycm-cmake-modules         0.16.2               h59595ed_0    conda-forge
zlib                      1.2.13               hd590300_5    conda-forge
zstandard                 0.22.0          py310h1275a96_0    conda-forge
zstd                      1.5.5                hfc55251_0    conda-forge

rcone11, 2024-05-07:
> Please read https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax on how to correctly format github comments (in particular use `~~~` blocks to contains the `conda list` output, thanks.

I'm sorry for my ignorance and I didn't  successfully compiled the /home/hablockfactory/example/matlab/BlockFactory.mexa64 file .

~~~
(base) han@han-System-Product-Name:~$ conda list
# packages in environment at /home/han/miniforge3:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
ace                        7.1.3                h59595ed_2    conda-forge
alsa-lib                  1.2.11               hd590300_1    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
aom                       3.8.2                h59595ed_0    conda-forge
archspec                  0.2.3              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                h8343317_3    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
binutils                  2.40                 h4852527_0    conda-forge
binutils_impl_linux-64    2.40                 hf600244_0    conda-forge
binutils_linux-64         2.40                 hdade7a5_3    conda-forge
blockfactory              0.8.5              h59595ed_120    robotology
boltons                   24.0.0             pyhd8ed1ab_0    conda-forge
brotli-python             1.1.0           py310hc6cd4ac_1    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
c-ares                    1.28.1               hd590300_0    conda-forge
c-compiler                1.7.0                hd590300_0    conda-forge
ca-certificates           2024.2.2             hbcca054_0    conda-forge
cairo                     1.18.0               h3faef2a_0    conda-forge
casadi                    3.6.5           py310h60ac289_0    conda-forge
casadi-matlab-bindings    3.6.3.1            h59595ed_120    robotology
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py310h2fee648_0    conda-forge
charset-normalizer        3.3.2              pyhd8ed1ab_0    conda-forge
cmake                     3.29.2               hcfe8598_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
compilers                 1.7.0                ha770c72_0    conda-forge
conda                     24.3.0          py310hff52083_0    conda-forge
conda-libmamba-solver     24.1.0             pyhd8ed1ab_0    conda-forge
conda-package-handling    2.2.0              pyh38be061_0    conda-forge
conda-package-streaming   0.9.0              pyhd8ed1ab_0    conda-forge
cxx-compiler              1.7.0                h00ab1b0_0    conda-forge
cython                    0.29.37         py310hc6cd4ac_0    conda-forge
dav1d                     1.2.1                hd590300_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
distro                    1.9.0              pyhd8ed1ab_0    conda-forge
eigen                     3.4.0                h00ab1b0_0    conda-forge
expat                     2.6.2                h59595ed_0    conda-forge
ffmpeg                    6.1.1           gpl_hee4b679_108    conda-forge
fmt                       10.2.1               h00ab1b0_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_1    conda-forge
fontconfig                2.14.2               h14ed4e7_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fortran-compiler          1.7.0                heb67821_0    conda-forge
freeglut                  3.2.2                hac7e632_2    conda-forge
freetype                  2.12.1               h267a509_2    conda-forge
fribidi                   1.0.10               h36c2ea0_0    conda-forge
gcc                       12.3.0               h915e2ae_6    conda-forge
gcc_impl_linux-64         12.3.0               h1562d66_6    conda-forge
gcc_linux-64              12.3.0               h6477408_3    conda-forge
gettext                   0.22.5               h59595ed_2    conda-forge
gettext-tools             0.22.5               h59595ed_2    conda-forge
gfortran                  12.3.0               h915e2ae_6    conda-forge
gfortran_impl_linux-64    12.3.0               h6d6b2fb_6    conda-forge
gfortran_linux-64         12.3.0               h617cb40_3    conda-forge
glfw                      3.4                  hd590300_0    conda-forge
[... 213 lines cut ...]
xkeyboard-config          2.41                 hd590300_0    conda-forge
xorg-fixesproto           5.0               h7f98852_1002    conda-forge
xorg-inputproto           2.3.2             h7f98852_1002    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libice               1.1.1                hd590300_0    conda-forge
[... cut ...]
xz                        5.2.6                h166bdaf_0    conda-forge
yaml-cpp                  0.8.0                h59595ed_0    conda-forge
yarp-matlab-bindings      3.9.1              h59595ed_120    robotology
ycm-cmake-modules         0.16.2               h59595ed_0    conda-forge
zlib                      1.2.13               hd590300_5    conda-forge
zstandard                 0.22.0          py310h1275a96_0    conda-forge
zstd                      1.5.5                hfc55251_0    conda-forge

rcone11, 2024-05-08:
> Please read https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax on how to correctly format github comments (in particular use `~~~` blocks to contains the `conda list` output, thanks.

I'm sorry to bother you again sir, but could you help me to check my problems in your free time

traversaro, 2024-05-08:
If  compiling `BlockFactory.mexa64` did not succeeded, can you report which commands did you used to try to compile it and what are the errors you encountered?

rcone11, 2024-05-08:
> If compiling `BlockFactory.mexa64` did not succeeded, can you report which commands did you used to try to compile it and what are the errors you encountered?

sir, I compiled the BlockFactory.mexa64 file by Matlab and the error is "MEX level2 S-function "BlockFactory" must be called with at least 4 input arguments."
when I run the model.mdl here is the error displayed.
~~~
Error while obtaining sizes from MEX S-function 'BlockFactory in "odel/w// mask'Caused by:
MEX file "/home/han/blockfactory/example/matlab/BlockFactory.mexa64' invalid:libBlockFactoryCore.so.0.8.5: cannot open shared object file : No such file or directory
Component:Simulink | Category:Model error
Error while obtaining sizes from MEX S-function 'BlockFacto!y in 'Model/w//o mask'.Caused by:
MEX file '/home/han/blockfactory/example/matlab/BlockFactory.mexa64' invalid:libBleckFactoryCore.so.0.8.5: cannot open shared object file : No such file or directory.
~~~

traversaro, 2024-05-08:
> sir, I compiled the BlockFactory.mexa64 file by Matlab and the error is "MEX level2 S-function "BlockFactory" must be called with at least 4 input arguments."

Which command(s) did you use to compile the `BlockFactory.mexa64` file?

rcone11, 2024-05-08:
> Which command(s) did you use to compile the `BlockFactory.mexa64` file?

I run files by right-clicking with the mouse and then selecting "Run".

traversaro, 2024-05-08:
Sorry, I am not sure what which files you mean and in which program are you right-clicking them. Sorry for the direct question,  are you familiar with what "compilation" means in C/C++?

rcone11, 2024-05-08:
> Sorry, I am not sure what which files you mean and in which program are you right-clicking them. Sorry for the direct question, are you familiar with what "compilation" means in C/C++?

I run BlockFactory.mexa64 files by right-clicking with the mouse and then selecting "Run" in Matlab.

traversaro, 2024-05-08:
You can't run `BlockFactory.mexa64` by itself, the `BlockFactory.mexa64` is meant to be insert in an existing Simulink model and run that one. However, I am still confused with the "I didn't compiled the BlockFactory.mexa64 succeed" sentence.

Can you answer to the question:

> Sorry for the direct question, are you familiar with what "compilation" means in C/C++?

Thanks!

rcone11, 2024-05-08:
> You can't run by itself, the is meant to be insert in an existing Simulink model and run that one. However, I am still confused with the "I didn't compiled the BlockFactory.mexa64 succeed" sentence.`BlockFactory.mexa64``BlockFactory.mexa64`
> 
> Can you answer to the question:
> 
> > Sorry for the direct question, are you familiar with what "compilation" means in C/C++?
> 
> Thanks!

Sir according to my understanding, "compiling" refers to the process of translating source code written in C or C++ into machine-readable binary code.

rcone11, 2024-05-08:
> Ok, so if you have have the `BlockFactory.mexa64` file, you indeed compiled it, is that correct? How did you compiled it?

 I tried to run the model.mdl in the example folder but, some errors occurred here is the error displayed.
~~~
Error while obtaining sizes from MEX S-function 'BlockFactory in "odel/w// mask'Caused by:
MEX file "/home/han/blockfactory/example/matlab/BlockFactory.mexa64' invalid:libBlockFactoryCore.so.0.8.5: cannot open shared object file : No such file or directory
Component:Simulink | Category:Model error
Error while obtaining sizes from MEX S-function 'BlockFacto!y in 'Model/w//o mask'.Caused by:
MEX file '/home/han/blockfactory/example/matlab/BlockFactory.mexa64' invalid:libBleckFactoryCore.so.0.8.5: cannot open shared object file : No such file or directory.
~~~

traversaro, 2024-05-08:
Running the `model.mdl` is something done **after** you compiled the `BlockFactory.mexa64` file. I am asking how you compiled (i.e. created) the `BlockFactory.mexa64` file.

rcone11, 2024-05-08:
sorry sir, I'm now very confused about how should I do to compile the BlockFactory.mexa64  correctly.

rcone11, 2024-05-08:
> Running the `model.mdl` is something that needs to bedone **after** you compiled the `BlockFactory.mexa64` file. I am asking how you compiled (i.e. created) the `BlockFactory.mexa64` file.

sorry sir, I found `"BlockFactory.mexa64"`, I'm now very confused about how should I do to compile the BlockFactory.mexa64  correctly.

rcone11, 2024-05-08:
sorry sir, I found `BlockFactory.mexa64` in /home/han/miniforge3/mex/BlockFactory.mexa64, I'm now very confused about how should I do to compile a block correctly.

traversaro, 2024-05-08:
Sorry, I just read that you wrote that in the original comment:

> my BlockFactory.mexa64 file address is /home/han/miniforge3/mex/BlockFactory.mexa64.But my model.mdl file address is /home/han/blockfactory/example/matlab/model.mdl.To ensure that these two files [...]

My bad, I did not read it correctly. Anyhow, please delete the `BlockFactory.mexa64` file from `/home/han/blockfactory/example/matlab/` and follow the instructions in https://robotology.github.io/blockfactory/mkdocs/create_new_library/ to compile the example. Note that if you are on Windows, you will need to:
* Install Visual Studio with C++ support (you need to choose either VS2019 or VS2022)
* Install the conda packages required to do C++ development, that are `cmake pkg-config ninja make` and if you are using vs2019 `vs2019_win-64`, otherwise `vs2022_win-64`.

rcone11, 2024-05-08:
I have deleted the `BlockFactory.mexa64` file from ` /home/han/blockfactory/example/matlab/`,then I run the model.mdl but now but I got the following errors in Simulink:
~~~
Error:Error reported by S-function 'BlockFactory' in 'Model/w// mask': Failed to create factory Factory error (2): ExampleToolbox: cannot open shared object file: No such file or directory Failed to get factory object (className=SignalMath,libName=ExampleToolbox) Failed to get the factory 

Error:Error reported by S-function 'BlockFactory' in 'Model/w//o mask': Failed to create factory Factory error (2): ExampleToolbox: cannot open shared object file: No such file or directory Failed to get factory object (className=SignalMath,libName=ExampleToolbox) Failed to get the factory 
~~~
then I followed this command in #70 `blockfactory-exists ExampleToolbox SignalMath` to check, and the output:
~~~
SUCCESS: Block "SignalMath" found and loaded from plugin "ExampleToolbox".
~~~
In addition, my current path in Matlab is `/home/han/miniforge3/mex`, and the model.mdl is in `/home/han/blockfactory/example/matlab`, what should I do to solve the problem? thanks

traversaro, 2024-05-08:
As I mentioned before:

> follow the instructions in the "Compiling instructions" section https://robotology.github.io/blockfactory/mkdocs/create_new_library/ to compile the example

as you are in linux, the compilers can be installed via conda: 

~~~
conda install cmake pkg-config ninja make compilers
~~~

rcone11, 2024-05-08:
> As I mentioned before:
>
>
> as you are in linux, the compilers can be installed via conda:
>
> ```
> conda install cmake pkg-config ninja make compilers
> ```

yes professor,I follow the steps and I have executed this command.`conda install cmake pkg-config ninja make compilers`

rcone11, 2024-05-12:
dear professor,I still encounter this problem：
~~~
Error reported by S-function 'BlockFactory' in 'Model/w// mask':
Failed to create factory
Factory error (2): ExampleToolbox: cannot open shared object file: No such file or directory
Failed to get factory object (className=SignalMath,libName=ExampleToolbox)
Failed to get the factory
Component:Simulink | Category:Model error
Error reported by S-function 'BlockFactory' in 'Model/w//o mask':
Failed to create factory
Factory error (2): ExampleToolbox: cannot open shared object file: No such file or directory
Failed to get factory object (className=SignalMath,libName=ExampleToolbox)
Failed to get the factory
~~~
I read #70 try to solve my problem and I set
`export BLOCKFACTORY_PLUGIN_PATH=$BLOCKFACTORY_PLUGIN_PATH:/home/han/miniforge3/mex`
I enter the following code `blockfactory-exists ExampleToolbox SignalMath` and the output is `ERROR: Failed to get factory object (blockName=SignalMath,pluginName=ExampleTool`
I read #61 to chenck my  WB-Toolbox is installed correctly and BLOCKFACTORY_PLUGIN_PATH is set correctly this are the output 
~~~
(base) han@han-System-Product-Name:~$ blockfactory-exists WBToolbox ForwardKinematics
SUCCESS: Block "ForwardKinematics" found and loaded from plugin "WBToolbox".
(base) han@han-System-Product-Name:~$ echo $?
0
(base) han@han-System-Product-Name:~$  blockfactory-exists WBToolbox BlockThatDoesNotExists
ERROR: Failed to get factory object (blockName=BlockThatDoesNotExists,pluginName=WBToolbox)(base) han@han-System-Product-Name:~$ echo $?
1
~~~
so, now I want to know how should I do to run the model.mdl successful @traversaro

closed 2024-07-09 by rcone11