robotology/blockfactory #81: Failure in compiling autogenerated code for block with constant (?) input

traversaro, 2024-09-23:
In an internal project, we had a compilation failure for some autogenerated code that used WB-Toolbox, a a blockfactory based library. We fixed manually the code with this change:

~~~
// Inputs
 blockInfo->setInputPort(
{ 0, { 4, 4 }, blockfactory::core::Port::DataType::DOUBLE },
 static_cast<void*>(&tsid_controller_mdl_B.basepose[0]));
 blockInfo->setInputPort(
 { 1, { 1, 2 }, blockfactory::core::Port::DataType::DOUBLE },
static_cast<void*>(&tsid_controller_mdl_B.TmpSignalConversionAtSFunctionI[0]));
blockInfo->setInputPort(
{ 2, { 1, 6 }, blockfactory::core::Port::DataType::DOUBLE },
 static_cast<void*>(&tsid_controller_mdl_B.TmpSignalConversionAtSFunctio_i[0]));
blockInfo->setInputPort(
{ 3, { 1, 2 }, blockfactory::core::Port::DataType::DOUBLE },
static_cast<void*>(&tsid_controller_mdl_B.TmpUnitConversionAtjoints_velIn[0]));
blockInfo->setInputPort(
 { 4, { 1, 6 }, blockfactory::core::Port::DataType::DOUBLE },
+ static_cast<void*>(const_cast<double*>(&tsid_controller_mdl_ConstP.pooled4[0])));
- static_cast<void*>(&tsid_controller_mdl_ConstP.pooled4])));
blockInfo->setInputPort(
{ 5, { 1, 2 }, blockfactory::core::Port::DataType::DOUBLE },
static_cast<void*>(&tsid_controller_mdl_B.Gain[0]));
~~~

My understanding is that the block for which code was generated had a constant input, and the code in https://github.com/robotology/blockfactory/blob/04e2ae2acc9298f8ea7601f2bb1ff797cb837778/matlab/BlockFactory.tlc#L203-L208 needs to be modified to account for this case. 

@mfussi66 @Gio-ds @DavideBarbone  can you share more info on the failure (for example more of the generated code, so it is possible to understand which specific block of WB-Toolbox is failing, and so that we can debug the 4th input to it that is probably "special" somehow?).

traversaro, 2024-09-23:
Or perhaps the issue is just on the blockfactory side of some missing `const` ?

traversaro, 2024-09-23:
> Or perhaps the issue is just on the blockfactory side of some missing `const` ?

This is possible, but the situation is a bit complicated as the whole library does not distinguish between non-const and const Signal, and eventually all constness information is stripped in https://github.com/robotology/blockfactory/blob/04e2ae2acc9298f8ea7601f2bb1ff797cb837778/sources/Core/src/Signal.cpp#L131 . Probably we can do a quick fix by just doing a const_cast in https://github.com/robotology/blockfactory/blob/04e2ae2acc9298f8ea7601f2bb1ff797cb837778/matlab/BlockFactory.tlc#L203-L208 .

traversaro, 2024-09-24:
Yes, I looked into this and it seems that there is a bit of confusion:

* [`CoderBlockInformation::setInputPort(const core::Port::Info& portInfo, void* signalAddress)`](https://github.com/robotology/blockfactory/blob/04e2ae2acc9298f8ea7601f2bb1ff797cb837778/sources/SimulinkCoder/src/CoderBlockInformation.cpp#L372) takes in input **a non-const pointer**, so it fails if you try to pass a const pointer
* `CoderBlockInformation::setInputPort` calls [`CoderBlockInformation::impl::storePortInfo(const core::Port::Info& portInfo, void* signalAddress, IndexToPortAndSignalDataMap& dataMap)`](https://github.com/robotology/blockfactory/blob/04e2ae2acc9298f8ea7601f2bb1ff797cb837778/sources/SimulinkCoder/src/CoderBlockInformation.cpp#L309), that again takes in input **a non-const pointer**
* [`CoderBlockInformation::impl::storePortInfo`] [calls](https://github.com/robotology/blockfactory/blob/04e2ae2acc9298f8ea7601f2bb1ff797cb837778/sources/SimulinkCoder/src/CoderBlockInformation.cpp#L360) [`Signal::initializeBufferFromContiguousZeroCopy(const void* buffer, size_t len)`](https://github.com/robotology/blockfactory/blob/04e2ae2acc9298f8ea7601f2bb1ff797cb837778/sources/Core/src/Signal.cpp#L109), that indeed finally takes in input a **const pointer**.
* `initializeBufferFromContiguousZeroCopy` uses this const pointer and it cast the constness away via `const_cast` in the line [`m_bufferPtr = const_cast<void*>(buffer);`](https://github.com/robotology/blockfactory/blob/04e2ae2acc9298f8ea7601f2bb1ff797cb837778/sources/Core/src/Signal.cpp#L131)

So, the whole call stack is a mixture of const and non-const, and furthermore they are the same data structure for both input and output ports, while for properly accounting const correctness we should have two different `Signal` structure, one for which const signals and another for non-const signals.

traversaro, 2024-09-24:
> So, the whole call stack is a mixture of const and non-const, and furthermore they are the same data structure for both input and output ports, while for properly accounting const correctness we [...]

Ok, this may be complex, so let's separate the concern:
* To quickly fix this issue, we can just add a `const_cast` (assuming that then we will never write inside a buffer of an input port
* For the proper solution, we open a dedicated issue: https://github.com/robotology/blockfactory/issues/83

mfussi66, 2024-09-26:
Code generation was successful:
[image]

I'm going to try to compile.

mfussi66, 2024-09-26:
Code compiles successfully :+1:

traversaro, 2024-09-26:
> Code compiles successfully 👍

Ok, can you approve https://github.com/robotology/blockfactory/pull/85 ? Thanks!

closed 2024-09-26 by traversaro